rg35xx-cfw/rg35xx-cfw.github.io #1: Initial RG35XX Issues

acmeplus, 2023-03-14:
Here's an initial set of issues on the GPU enabled batocera and retroarch firmwares for the RG35XX:

General issues:
- [ ] Power button is not currently working. It works after enabling triggerhappy, but the actual shutdown is not a full shutdown.
- [ ] Volume keys not responding. Events are properly recognized, but they are not being captured by triggerhappy correctly, therefore not sending the volume +/- changes. For retroarch there's a workaround, use the hotkey (menu key) + Volume keys
- [ ] Brightness control is not working
- [ ] Headphone detection does not work
- [ ] HDMI video out works in mirroring only if the HDMI cable is connected before booting
- [ ] HDMI audio does not work
- [ ] Some combination of cores+roms may not work due to the system’s low memory. ram swap has been enabled but there are still issues
- [ ] Text entry in retroarch and batocera is currently broken since all input codes are letters and there are no modifiers

Batocera issues:
- [ ] Batocera-resolution needs to be fixed: xandr version mismatch does not provide the required command line options. Resolution needs to be forced to 640x480 for internal and 1280x720 for HDMI
- [ ] emulator launcher does not add the proper controls.
- [ ] Batocera-audio still to be done

8trius, 2023-03-14:
I'm new to Github, so I'm happy to receive any correction in how to file bug reports.

After installing on my pre v6 motherboard unit (using a 16GB SanDisk MicroSD card), here's what I'm experiencing:

1. A consistent humming noise coming from the speaker. It almost sounds like a spinning hard drive sound. This occurs even when emulation sound is reduced to the absolute lowest of -80db.
2. Once I press Menu + R2 to frame advance, I can't figure out how to resume normal speed emulation.
3. For Golden Sun (for the GBA), I have tried both mGBA and gbSP. In mGBA, when attacking an enemy with a normal attack, there are audio stutters.
4. For Mario Golf: Advance Tour, there are significant audio stutters as well as frame drops for the overhead view as well as the animation for the golf ball being hit and sent across the screen.
5. For Kirby's Dream Land, I've tried Gambette and mGBA and both of them produce inconsistent frame rate stutters. The frame rate is not locked to 60FPS, it will vary above and slightly below, depending on what's going on. Same for Metroid: Zero Mission on mGBA.

rogerillu, 2023-03-14:
First of all, this is aswesome. Keep up the good work. It's smothly and for an alpha version is very playable. 

- SNES: Super Mario World is working perfect
- PSX: Crash Bandicoot 2 is working perfect with rearmed emulator
- Dreamcast: Crazy taxi loading but not playable, very slow and need to remap buttons, but very slow
- Nintendo 64: Tried SuperMario64 and Goldeneye. Not working and both cores mupen and paralell crash :(

I understand if it loads snes games, it will do the same with megadrive, mastersystem, gameboy, etc. 

And yes, when starting there's a humming noise. If you restart multiple times sometimes it disappear.

jamezh2020, 2023-03-14:
I’m really looking forward to this release 😊 It’s awesome to see what will be possible for this device
Issues:
All of the above plus I’m experiencing random system shutdown's or halts.

J3rUr0, 2023-03-14:
In my initial tests I don't get a smooth experience on the MD,SFC cores. 

If you try 240p test suite ->Video Scroll Test you will notice that there is a problem with tearing / stuttering. That problem also occurs in GarlicOS and MinUI but is much less noticeable. 

Black-Seraph said on twitter that the problem is that we don't know the exact screen specs / refresh rate. In GarlicOs we can disable Audio Sync and we get a perfect smooth picture (with some cons.) but I can't get it to work here.

JonSRP, 2023-03-15:
My RG35XX PCB version is a V6.0 (dated 2022-11-19) 
The haptic feedback motor starts on boot (something loading in the kernel.dtb?) 
It won't stop, I tried resetting a bunch of times no dice.
It's too annoying to hold to test any further.

acmeplus, 2023-03-15:
> My RG35XX PCB version is a V6.0 (dated 2022-11-19)
> The haptic feedback motor starts on boot (something loading in the kernel.dtb?)
> It won't stop, I tried resetting a bunch of times no dice.
> It's too annoying to hold to test any further.

Yes, it seems it mostly impacting V6 PCBs. Unfortunately I don't have one so it's difficult to troubleshoot. Try this kernel.dtb and let me know if it improve things. 
[kernel.dtb.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/10979947/kernel.dtb.zip)

acmeplus, 2023-03-15:
> 1. A consistent humming noise coming from the speaker. It almost sounds like a spinning hard drive sound. This occurs even when emulation sound is reduced to the absolute lowest of -80db. UPDATE: It actually feels like the vibe motor is engaging. I performed three resets in a row and it finally just stopped.

Yes, the issue seems to be more persistent in V6 motherboards. Ticket is not here #2 

> 2. Once I press Menu + R2 to frame advance, I can't figure out how to resume normal speed emulation.

START+SELECT by default returns to the retroarch menu, from there you can redefine the hotkeys. I don't remember what  Menu+R2 is mapped to but you can reassigned to toggle instead of switch.

> 3. For Golden Sun (for the GBA), I have tried both mGBA and gbSP. In mGBA, when attacking an enemy with a normal attack, there are audio stutters.
> 4. For Mario Golf: Advance Tour, there are [...]

See the latest comments from BlackSeraph on discord, there may be an issue with the gpu governor and the CPU speed. I'm going to open a separate ticket to track those issues.

acmeplus, 2023-03-15:
> * Dreamcast: Crazy taxi loading but not playable, very slow and need to remap buttons, but very slow

This is expected, this device is not powerful enough, and while some games run at "playable" speeds most of the DC catalogue is out of reach.

> * Nintendo 64: Tried SuperMario64 and Goldeneye. Not working and both cores mupen and paralell crash :(

There's an issue with memory allocation and N64 cores. I can get them to run via command line without loading retroarch frontend, but not once the menu is up. I'll open a ticket just to track down N64 compatibility. 

> And yes, when starting there's a humming noise. If you restart multiple times sometimes it disappear.

Do you know which motherboard version do you have? Thanks!

rogerillu, 2023-03-15:
> This is expected, this device is not powerful enough, and while some games run at "playable" speeds most of the DC catalogue is out of reach.
>
>
> There's an issue with memory allocation and N64 cores. [...]

It would be nice to have N64 emualtor running.

Where can I check the motherbord version? I cannot see the info in the System Information.

acmeplus, 2023-03-15:
> It does not, my RG35XX is still vibing - starts at about 4/5 seconds after pressing reset button.

Thanks for checking. It's really frustrating because I have other V6 users that don't see the issue at all, or ones that only until they reset it, so I suspect not all V6 are the same... I'll be using this ticket #2 for follow ups and test updates.

acmeplus, 2023-03-15:
> It would be nice to have N64 emualtor running.

Yes, will be addressed once I figure out the other major issues.
 
> Where can I check the motherbord version? I cannot see the info in the System Information.

It's internal, no need to open the console up unless you really want. It seems the issue is not so linked to the actual version.

acmeplus, 2023-03-16:
> In my initial tests I don't get a smooth experience on the MD,SFC cores.
>
> If you try 240p test suite ->Video Scroll Test you will notice that there is a problem with tearing / stuttering. That [...]

I noticed a stutter every second or so, but it seems to me it's related to a periodic process and not to the actual screen refresh rate. That's not to say that a similar behavior could be due to the screen refresh in Garlic/Stock, since those use framebuffer rendering. So I suspect the issue is different here. In any case there's limited room for improvement since the GPU drivers are closed sourced. Right not the GPU is working on top of X11, if at some point I manage to get it work directly with KMSDRM we may see some improved performance, but that really depends on having a PowerVR module for libdrm.

jamezh2020, 2023-03-18:
Maybe it’s just me but what file dinings e to create to boot into RA? Text file?

acmeplus, 2023-03-18:
Yes. A simple text file. The only important content is the filename that needs to be called startRA.

jamezh2020, 2023-03-18:
> Yes. A simple text file. The only important content is the filename that needs to be called startRA.

Okay, then I got I right in the first place. But when I put it in the root of the second SD card I’m booting straight into batocera.

David-Martin-Pereira, 2023-03-20:
I am having an issue navigating through my N64 catalogue in batocera. Not starting any game, but just browsing, I get caught in a reboot loop.

Configuration:
- Single SD card 64 gB. I resize the 2nd partition to have space for my roms.
- I used the suggested fix of replacing kernel.dtb to succesfully eliminate the motor issue.
- I placed my bios under /userdata/bios as suggested by the batocera file structure here: https://wiki.batocera.org/add_games_bios

Also, the bios don't seem to be found by the either batocera nor retroarch. Is there a particular format it is expecting (uppercase or smallcase I tried both for psxonpsp660.bin in psx with no success)?

acmeplus, 2023-03-20:
> I am having an issue navigating through my N64 catalogue in batocera. Not starting any game, but just browsing, I get caught in a reboot loop.
> 

The current version does not have enough memory to launch N64 roms via the menu. That will be fixed in the next release. 

> Also, the bios don't seem to be found by the either batocera nor retroarch. Is there a particular format it is expecting (uppercase or smallcase I tried both for psxonpsp660.bin in psx with no success)?

The first time you start a game via retroarch and set the bios path in settings, directories, system path.

rogerillu, 2023-03-20:
> It's internal, no need to open the console up unless you really want. It seems the issue is not so linked to the actual version.
> 

RG35XX_V6.0 2022-11-19

EBLeifEricson, 2023-03-25:
This is kind of a minor thing so not sure if it's worth it's own issue yet, but the [official pico-8 engine](https://wiki.batocera.org/systems:pico8) seems to not work. 

Already did some digging trying to get it to work on GarlicOS. Unfortunately the only ARM build is the Pi version, which relies on a Pi-specific graphics library (bcm_host), so I suspect it's related to that-- in which case not sure if anything can be done about it.

There are some carts that don't work with retro8 which is why I mention it.

acmeplus, 2023-03-25:
> This is kind of a minor thing so not sure if it's worth it's own issue yet, but the [official pico-8 engine](https://wiki.batocera.org/systems:pico8) seems to not work.
>
> Already did some digging [...]

The system supports pico8 via emulation with retro8 core, or you can use the official lexaoffle RPI dynamic version but that still requires some configuration and has not been tested in this latest version. You can use pico8_dyn, but at this point keys are not going to work. I've just tracked this into a new ticket #13

EBLeifEricson, 2023-03-25:
Great, thanks. Some carts just simply don't work/are broken with retro8 so have the Lexaloffle engine would be amazing.

kitoymonster, 2023-03-25:
Noob questions from someone who just heard/used batocera. I already set the controllers and assigned it to P1 but 
1. I cannot exit the sample 2048 game by pressing the menu which I assigned as hotkey. The only way I could exit is by pressing the reset key.
2. I couldn't load my rom files in the slot2 even if I used the stock OS with roms in it

MauricioSuporte, 2023-04-16:
Have you heard about the language switch not working? I switched to Portugues (Brasil) and even restarting and showing that it's already switched when I open the settings menu, the texts haven't changed.

Jehmii, 2023-04-22:
v0.7 has gone back to being stuck on boot screen with v6 revision 

actually v0.6 didn’t work either only the test v0.6 that was posted for the new revision boards.

Any ideas?
Flashed the card brand new with 0.7
Stuck at boot

acmeplus, 2023-04-22:
> v0.7 has gone back to being stuck on boot screen with v6 revision
> 
> actually v0.6 didn’t work either only the test v0.6 that was posted for the new revision boards.
> 
> Any ideas? Flashed the card brand new with 0.7 Stuck at boot

Let me double check that there was not a regression. It seems like V6.0 are definitely different in terms of boot sequence. However it does not affect to all v6.0 models.

Jehmii, 2023-04-22:
Just tested v0.6 (not test build) single SD and that booted

rested v0.7 again, doesn’t boot.

also tried to just flash v0.6 and update to v0.7 but not enough room on first partition to copy the boot archive.

acmeplus, 2023-04-22:
> also tried to just flash v0.6 and update to v0.7 but not enough room on first partition to copy the boot archive.

You need to delete boot/batocera first, then you should have space to update.

Jehmii, 2023-04-22:
Tried update from v0.6 to v0.7
Stuck at boot

acmeplus, 2023-04-22:
> Tried update from v0.6 to v0.7
> Stuck at boot

At this point the best option for cases like that is to use 0.6 and update the first partition files with boot.tar.xz from 0.7. You may need to delete boot/batocera to have enough space

Jehmii, 2023-04-22:
That’s what I did
Flashed v0.6, it boots.
Extracted boot.tar.xz for v0.7 
Deleted batocera file from first partition 
Copied boot.tar.xz files from v0.7 to first partition
Stuck at boot screen

acmeplus, 2023-04-22:
> That’s what I did
> Flashed v0.6, it boots.
> Extracted boot.tar.xz for v0.7
> Deleted batocera file from first partition
> Copied boot.tar.xz files from v0.7 to first partition
> Stuck at boot screen

Ok. One more thing to try. Copy first the kernel.dtb from v0.6 and try to boot. If that does not work copy now the uImage from 0.6. 

Let’s see if one of those steps solves the issue. Thanks!

Jehmii, 2023-04-22:
Updated to v0.7
Copied kernel.dtb from v0.6
Boots past boot screen but stuck on black screen

copied uimage from v0.6 
Boots! Success!

acmeplus, 2023-04-29:
HDMI audio output is working in v0.8. Go to Settings -> System Settings -> Audio Output and change it to the bottom option