rg35xx-cfw/rg35xx-cfw.github.io #2: Haptic engine is engaged all times. In some cases resetting works as a temporary workaround, but V6 motherboards users seem to be unable to disable it.

acmeplus, 2023-03-15:


8trius, 2023-03-15:
Yes, I just opened up my device to confirm: I'm running a v.6.0 2022-11-19 motherboard and this issue is persistent. Three resets in a row stopped it. But once I switch to another OS like GarlicOS, or reset, the issue can reactivate.

JonSRP, 2023-03-17:
In the Test Release Alpha 0.2 I was finally able to get mine to stop by using suspend in the Batocera QUIT GUI. When resuming I hear a pop from the speaker and it rebooted with the Haptic engine/motor disabled. subsequent reboots/shutdowns it seems to stay disabled too.

JonSRP, 2023-03-17:
Well it came back but after lots of trial and error I think I found the problem..... I removed this from the DTS 

`vdd-gpu-dcdc {		
				compatible = "actions,atc2603c-ext-pwm-dcdc1\0actions,atc2603c-ext-pwm-dcdc";		
				regulator-name = "vdd-gpu-dcdc";		
				regulator-min-microvolt = <0xdbba0>;		
				regulator-max-microvolt = <0x1312d0>;		
				regulator-always-on;		
				pwms = <0x0d 0x01 0xa6a 0x00>;		
				table_len = <0x0f>;		
				vdd_cpu_table = <0xdbba0 0xe1d48 0xe7ef0 0xee098 0xf4240 0xfa3e8 0x100590 0x106738 0x10c8e0 0x112a88 0x118c30 0x11edd8 0x124f80 0x12b128 0x1312d0>;		
				pwm_config_table = <0x13 0x12 0x11 0x10 0x0f 0x0e 0x0d 0x0c 0x0b 0x0a 0x09 0x08 0x07 0x06 0x05>;		
				linux,phandle = <0x1b>;		
				phandle = <0x1b>;		
			};		
			vdd-core-dcdc {		
				compatible = "actions,atc2603c-ext-pwm-dcdc2\0actions,atc2603c-ext-pwm-dcdc";		
				regulator-name = "vdd-core-dcdc";		
				regulator-min-microvolt = <0xdbba0>;		
				regulator-max-microvolt = <0x1312d0>;		
				regulator-always-on;		
				pwms = <0x0d 0x02 0xa6a 0x00>;		
				table_len = <0x0f>;		
				vdd_cpu_table = <0xdbba0 0xe1d48 0xe7ef0 0xee098 0xf4240 0xfa3e8 0x100590 0x106738 0x10c8e0 0x112a88 0x118c30 0x11edd8 0x124f80 0x12b128 0x1312d1>;		
				pwm_config_table = <0x13 0x12 0x11 0x10 0x0f 0x0e 0x0d 0x0c 0x0b 0x0a 0x09 0x08 0x07 0x06 0x05>;		
			};`

converted back to dtb and it booted without the haptic vibes. 

[kernel.dtb.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/11001097/kernel.dtb.zip)

acmeplus, 2023-03-17:
> Well it came back but after lots of trial and error I think I found the problem..... I removed this from the DTS
[...]
> converted back to dtb and it booted without the haptic vibes.
> 
> [kernel.dtb.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/11001097/kernel.dtb.zip)

Very interesting, supposedly the gpu needs that block, but just checked and as far as the gpu is enabled in its gpu section below it seems to work. 

I've narrowing down the issue to a conflict in the pwm. In theory that block should just enable the GPU, but it's possible that there's an overlapping with the motor pwm.

I've sent a modified file to some testers to see if it works. In my rg35xx the motor stays off most of the time.

JonSRP, 2023-03-17:
what's even more interesting is using this kernel on garlic os and changing video_driver = "gl" in the retroarch.cfg is that it boots up and shows using gl in the output now...

fouadchamoun, 2023-03-17:
I flashed the v0.2 alpha and had the haptic issue. I then replaced the kernel.dtb file with the modified one and it fixed the issue for me.

JonSRP, 2023-03-17:
This is a better device tree blob I think

I modified latest stock and only changed the GPU section to okay rather than disabled. It seems to boot into Batocera okay and as a bonus headphones now work :) 
 

`	gpu@b0300000 {
		compatible = "actions,atm7039c-gpu";
		reg = <0xb0300000 0x10000>;
		interrupts = <0x00 0x03 0x04>;
		normal_value = <0x02>;
		status = "okay";
	};`

[kernel.dtb.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/11005000/kernel.dtb.zip)

acmeplus, 2023-03-23:
This is now fixed in the upcoming release. Closing this issue.

Thanks @ghost for the hint.

closed 2023-03-23 by acmeplus