rg35xx-cfw/rg35xx-cfw.github.io #6: Source code?

lowlydba, 2023-03-17:
Since this is based off [Batocera Linux](https://github.com/batocera-linux/batocera.linux) which has a [GPL2 license](https://choosealicense.com/licenses/gpl-2.0/), can we expect there to be source code with GPL2 licensing released at some point?

acmeplus, 2023-03-17:
> Since this is based off [Batocera Linux](https://github.com/batocera-linux/batocera.linux) which has a [GPL2 license](https://choosealicense.com/licenses/gpl-2.0/), can we expect there to be source code with GPL2 licensing released at some point?

Of course, most of the code is already in the repos here. Of course the binary blobs are a different matter, but will upload them in binary form. 

Right now it uses the latest batocera for its build for the emulators, but it's not build from batocera but from buildroot.

Right now the buildroot and kernel modifications are in this project (https://github.com/rg35xx-cfw). The batocera repo is just missing the changes because I'm still stabilizing things.

The gpu blobs will be here but you can locate them in the cubieboard6 ftp.

acmeplus, 2023-03-17:
> Awesome, thanks!
> 
> > The batocera repo is just missing the changes because I'm still stabilizing things.
> 
> I think this is what was confusing initially :)

Yeah, in general it uses the latest batocera repo as-is, the only major thing is the two patches in batocera-emulationstation have been removed because the rest of the SDL2 joystick patches are not included in the main build. The other changes I believe are really minor. However there will be more significant patches once I switch the build to be done from batocera and not from buildroot.

closed 2023-03-17 by lowlydba