rg35xx-cfw/rg35xx-cfw.github.io #8: Charging only works while power is on

acmeplus, 2023-03-23:


acmeplus, 2023-04-22:
Partially fixed in v0.7. Opening a new issues for implementation fixes

soaresluciano, 2023-05-28:
I'm on v0.81 and every time I connect the power cable I see a splash screen with something like "DenX software engineering", and then it proceeds, to what I assume, to be the regular Batocera boot process. Is that the expected behavior?
Please let me know if you need more information.

robbo247, 2023-05-28:
The only firmware I have seen that doesn’t do it is the original firmware ,the bob firmware I am using at the minute also does the same and turns the unit on b4 charging but gets hot doing it but this is the best firmware I have tried yet  Sent from Mail for Windows From: Luciano SoaresSent: 28 May 2023 19:37To: rg35xx-cfw/rg35xx-cfw.github.ioCc: SubscribedSubject: Re: [rg35xx-cfw/rg35xx-cfw.github.io] Charging only works while power is on (Issue #8) I'm on v0.81 and every time I connect the power cable I see a splash screen with something like "DenX software engineering", and then it proceeds, to what I assume, to be the regular Batocera boot process. Is that the expected behavior?Please let me know if you need more information.

EvaPilot1, 2023-06-06:
I just wanted to say thank you for your hard work.

I think this is the only big issue I am currently having with this firmware. You can't plug in the device to charge it without it turning on, and if you try to turn it off, it turns back on again.

I've also noticed that while it's charging, you start to get audio pops and low level noise from the speaker. I thought it was the device struggling to play video in the menus, but realised the noise went away when it wasn't plugged in.

Also another extremely minor thing. The system update function doesn't recognise wifi. Not like there is any need for it right now, but I think it would be very cool in the future to update the device over WiFi.

scottchiefbaker, 2023-06-20:
This is my biggest issue... I plug in to charge and it boots up and plays the menu music on loop.

acmeplus, 2023-06-20:
> This is my biggest issue... I plug in to charge and it boots up and plays the menu music on loop.

Try [koriki](https://github.com/rg35xx-cfw/Koriki/releases/tag/koriki_rg35xx_v0.8) instead where the charging behavior has been improved. Batocera won't be updated until koriki reaches 1.0, but most of the issues in batocera are fixed in koriki (because it's the same foundation).

ThePenultimateNinja, 2023-09-29:
I have heard that Batocera on the RG35XX has been abandoned in favor of Koriki, but you said above that you are planning to update Batocera when Koriki reaches 1.0.
Is that still the case?

ThePenultimateNinja, 2023-09-29:
That's excellent news, I'm so glad it hasn't been abandoned. Koriki is fantastic too, but I just love Batocera.

GeckoPCB, 2023-10-27:
My first RG35XX has a problem of ALWAYS HIGH on the BATFET_EN pin as shown below.
The solution is shorting the GATE of PFET to the near GND (please check the 2nd image).
[image]

[image]

Xerxes0198, 2024-07-17:
@GeckoPCB thanks for the tip! Was this a permanent short, or just a once off?

GeckoPCB, 2024-07-17:
> @GeckoPCB thanks for the tip! Was this a permanent short, or just a once off?

A permanent short.

Xerxes0198, 2024-07-17:
> > @GeckoPCB thanks for the tip! Was this a permanent short, or just a once off?
> 
> A permanent short.

Thanks! I will give this a shot, I have nothing else to lose.