rg35xx-cfw/rg35xx-cfw.github.io #12: alpha 0.3 amiga, n64 showing 1/4 of the screen

grgohr, 2023-03-25:
tested n64 (libretro) and amiga1200 rom , they both showed 1/4 of the screen without any sound, and no way to exit without resetting

acmeplus, 2023-03-25:
> tested n64 (libretro) and amiga1200 rom , they both showed 1/4 of the screen without any sound, and no way to exit without resetting (bios folder not recognized so may be related)

Copy the retroarch.cfg file from the release page into system/.config/retroarch.cfg. 

That typically happens if you either use a second sdcard or if you have reformatted the second partition, so the image retroarch.cfg is lost.

grgohr, 2023-03-25:
thanks, missed that one, solved the issue.

I can't still figure out where to put the bios folder on the 2nd sdcard, in the retroarch.cfg file it shows 
system_directory = "~/roms" but does not work either

acmeplus, 2023-03-25:
> thanks, missed that one, solved the issue.
> 
> I can't still figure out where to put the bios folder on the 2nd sdcard, in the retroarch.cfg file it shows system_directory = "~/roms" but does not work either

That’s probably an error on my side. It should point to either ~/roms/bios or /userdata/system/roms/bios

rogerillu, 2023-03-26:
How can you load n64 roms?  I tried with mupen and parallel with no luck. It returns to the main menu.

Btw, the SHARE partition is only readable from linux, not from windows.

Thanks!!

grgohr, 2023-03-27:
you need to test several times in a row after each crash, you can check here https://www.youtube.com/watch?v=iwcZfej0F2U

thanks I managed to deal with the SHARE through a VM