rg35xx-cfw/rg35xx-cfw.github.io #18: Completely Inoperable - Crashes as soon as you press anything

esmith13, 2023-04-10:
So I have a brand new RG35XX that I received March 31st. I have played with the stock OS on stock SD card - no issues. I have used GarlicOS on my own SanDisk SD card - no issues. I have tried both Alpha 0.3 and now Alpha 0.4 of Batocera (wiped SD card with SD Formatter between versions) and have the exact same issues with both releases.

Batocera sets up the sdcard, then boots fine to ES and the frontend music begins to play. If I touch nothing the music will play and change tracks as expected.
As soon as I touch ANY input (dpad, face button, trigger or VOL+/-) one of two things happens:
1) there is a loud momentary squelch/static sound and the screen goes completely black and the device resets (and the cycle repeats once it boots again)
2) The screen and audio freeze completely and the image glitches to a green static filled variation of what was previously displayed and nothing but holding power a long time or pressing reset will get out of the situation (for it to just reboot and repeat again)

I have literally not been able to open a menu, launch a game, or anything.
I have tried new SD cards, tried repurposing an SD card that worked fine with GarlicOS, tried a card as small as 16GB and as large as 256GB. I have gone back to messing with the stock OS and Garlic OS and have never had any audio/video/freezing issues at all so I can't imagine it's a hardware issue either.

Any Ideas? Any chance of a small hardware revision in my unit causing this?
I am far from new to retro game handhelds or batocera for that matter so I'm very confident I am not making any dumb user errors and that my SD card(s) and flashing of the image are not causing the issue.

If there is any chance this is a hardware revision related issue I'd be happy to open mine up and take pictures or do whatever else can help diagnose the issue.

Attached is a pic of the "green static freeze glitch"

[image]

acmeplus, 2023-04-10:
Do you know which revision of the board is that device? Also, do you know if the battery is charged? Can you repeat with the power connected?

esmith13, 2023-04-10:
Same issue with 5v,3a power connected. Battery was 84% already (barely visible in new static green pic attached as well as lightning bolt in battery icon)
Board is marked with "RG35XX_v6.0" and "2022-11-19" (also attached)
Stock OS reports FW version "V1.20221223" (also attached)

thanks for the reply. Let me know if you need/want any more info or pics

[image]
[image]
[image]

acmeplus, 2023-04-10:
Thanks, if you can upload the kernel.dtb from your stock card and I'll modify to enable the gpu so you can test it.

esmith13, 2023-04-10:
See attached. Thanks again for your time.

[kernel.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/11193732/kernel.zip)

esmith13, 2023-04-10:
The end result was the same issue, but I was able to navigate around in ES for about 30-45 seconds before it locked up, went green and screeched the audio. 
I was able to lower the volume with the +/- keys, use the dpad as well as A/B & START buttons. The moment it locked up I wasn't actually pressing anything, just reading the menu page i was on in ES.

acmeplus, 2023-04-10:
> The end result was the same issue, but I was able to navigate around in ES for about 30-45 seconds before it locked up, went green and screeched the audio.
> I was able to lower the volume with the [...]

Does your unit work well with garlic with ++ overclock?

esmith13, 2023-04-10:
Apparently it does not. I had never actually tried it before but it locks up instantly on heavy games and after about 10 to 20 seconds on lightweight games and occasionally won't even lock up on the main menu. The effect you see when it locks up is different but not different enough that I would not say it is absolutely the same thing going on.

acmeplus, 2023-04-11:
> just read up on the "screen fuzz" issues with garlicOS and I'm actually running the version that should prevent the issue from happening at all and it still happens... Guess I'm one of the biggest "silicon lottery looser" of them all. :(

Try to set the governor lower in batocera. Edit the /userdata/system/batocera.conf and set this variable ``system.cpu.governor=ondemand`` (instead of performance)

esmith13, 2023-04-11:
OK. I had it working without crashing for about 30min after changing the  governor setting to "ondemand" and now after that time and a couple reboots I am back to green fuzz crashing within 30sec after booting again?!? I verified the governor setting is still "ondemand". I don't know what else to check at this point...

acmeplus, 2023-04-11:
That seems like a thermal issue. Leave it powered off for a while and repeat the test. There are even lower frequencies you can set like ``conservative``

esmith13, 2023-04-11:
Switched to conservative and also fully charged the battery while powered off again using the stock OS sd card. Then I left it for the night unplugged and untouched. Booted batocera this morning and got about 5min of playtime out of the shareware GBA game included in the image and then got the green fuzz freeze again. Device itself was still cool to the touch on the rear in both the battery area and the shielded area between the triggers.

Let it sit ~20min completely off and then tried again with batocera. This time I got the green fuzz while still in ES - about 1min after it was fully booted. had not even tried to launch a game yet.

esmith13, 2023-04-11:
I understand but hopefully you would humor me for one more question...
Is there a way to verify while the system is running what governor you are using? I ask because not only do I not get any noticable results from changing that in the /userdata/system/batocera.conf file but I also tried changing two items in the "C - Audio" section and do not see my changes reflected in the running system.
I set:
audio.volume=30         (was blank)
audio.bgmusic=0       (was 1)

The system still boots to max volume audio and music playing.
Also, oddly enough I just noticed retroarch plays no audio in any core?
It also may be worth noting that it NEVER crashes when you're in a game - only when your in ES...
All of the above was true on 0.4 and is still the case on 0.5 (updated, not clean image)

esmith13, 2023-04-11:
Thanks for that.
I will keep trying as new version release but will refrain from posting to this issue unless it is magically fixed or you specifically mention proper working governor support in the changelog so as not to waste your time or make this thread any needlessly longer than it already is.

Many thanks again for taking the time to help me and of course for making this awesome port.   :-)

acmeplus, 2023-04-12:
@esmith13 there's a test version for the newer boards, you may want to give it a try: https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/releases/tag/rg35xx_batocera_lite_alpha_v0.5_6C_Test

esmith13, 2023-04-13:
Thank you for the heads up.
v0.5C_6C_Test worked extremely well on my Rev_6.0 board until the device got warm after about 30min of use in heavy games (such as Bloody Roar II - PS1) and then it went crazy as usual until it was allowed to cool. Clearly I am having thermal issues on top of the typical "silicon lottery" problems.
I submitted an RMA for my device and should have a new one today or tomorrow. 
All subsequent testing by me will be done on this new device and I will be sure to identify the board and list the info in my next reply.

acmeplus, 2023-04-13:
Thanks for the update. You may also want to test v0.6 and lower the frequency.

esmith13, 2023-04-14:
Well, my old unit was definitely defective in some way. My new one has the exact same revision and date on the PCB (V6.0 2022-11-19). It works great with both v0.5C and v0.6. Tested both of those with the governor set to both "ondemand" and "performance" and never had a single crash or glitch after over an hour of play, despite the device getting warm when used in "performance" mode.

As a side note (not a complaint, just an observation) - While "performance" runs noticeably faster than "ondemand" for games like Bloody Roar II on PS1, the game's framerate is roughly equivalent to GarlicOS running the same game in it's normal mode. "++" mode on GarlicOS plays the game with barely a hiccup. Is it safe to assume this is likely due to cores and such not being optimized for the RG35XX at this time?
If not obvious at this point I must have "won" the lottery this time. GarlicOS ++ Mode can be used for an extended period of time without any glitches or crashes as well.

I can't wait for Batocera on the RG35XX to mature. This OS on such a tiny device is pure magic!

acmeplus, 2023-04-14:
Happy to hear that. Cores are not optimized in any way or form, but you have to also take into account that garlic is highly optimized down to the display frequency, etc. The GPU leverages most of the work and is able to do many things without optimizations, but I'm not surprised.

Good data points for comparison though, I'm always looking for good A/B tests to compare.

I'm going to close this ticket, but feel free to add more comments if you have more findings, or open a new one completely for other things that you may find. Thanks for all the feedback.

closed 2023-04-15 by acmeplus