rg35xx-cfw/rg35xx-cfw.github.io #19: RA menu not currently accessible

neocvera, 2023-04-10:
I have not found a button combination that enters RA menu during gameplay. Although I have found the following: 

menu+A = restart game
menu+X = load
menu+Y= save
menu+start = exit to ES / menu+B = exit to ES / menu + L1 = exit to ES
menu+up = select save slot (increase #)
menu+down = select save slot (decrease #)
menu+(hold)right = FF (fast forward)

acmeplus, 2023-04-11:
Yes, it's an issue because v0.4 has an issue with GL content (incorrect compilation that I need to fix). As a workaround, set the retroarch menu to RGUI: In EmulationStation go to ``Settings -> system settings -> Frontend Developer Options -> Retroarch Menu Drive -> RGUI``

neocvera, 2023-04-11:
Thank you, yes - your workaround confirmed working. 
[image]

acmeplus, 2023-04-11:
Fixed in v0.5

acmeplus, 2023-04-12:
> Hello,
> Just tested the v0.5 and the RA menu seems not available with N64 games.
> But works ok with other systems, thank you.

Some standalone emulators don't have working menus. But the main issue was in RA and it's now fixed.

acmeplus, 2023-04-12:
Closing issue, fixed in v0.5

closed 2023-04-12 by acmeplus