rg35xx-cfw/rg35xx-cfw.github.io #23: Stuck on Boot Splash Screen Logo

EdwardTheBanker, 2023-04-11:
Hi i already tried burning with 2 micro sd one with 128gb and another with 256gb using Rufus, Balena Etche and Win32Diskimager but no success.
i don't know which version im at just that the battery capacity is 2600mah.
Stock micro sd works fine btw.

acmeplus, 2023-04-11:
Have you tried garlic. If so which version(s) and which one works for you?

EdwardTheBanker, 2023-04-11:
> Have you tried garlic. If so which version(s) and which one works for you?

I haven't tried any cfw besides batocera, but i will try rn.

EdwardTheBanker, 2023-04-11:
I just installed and got the same problem as batocera, stuck on boot splash screen logo.

acmeplus, 2023-04-12:
> I just installed and got the same problem as batocera, stuck on boot splash screen logo.

try the latest garlic 1.4.6

EdwardTheBanker, 2023-04-12:
I tried lastest Garlic and it booted 1st try...

EdwardTheBanker, 2023-04-12:
it fixed it's working know, ty very much.

EdwardTheBanker, 2023-04-12:
it's working now.

closed 2023-04-12 by EdwardTheBanker

Allexxo92, 2023-04-12:
@acmeplus its a new board on the rg35xx 6.0c

qcai99jjk, 2024-08-20:
hi i think i got the same problem, i open the case & the board written as rg35xx_V6S  2023-06-07 . Please see the picture below. can you help me with boot issue, its just went blank with green light led 
[image]