rg35xx-cfw/rg35xx-cfw.github.io #25: Workaround to launch Drastic standalone emulator

TsukiNoKnight, 2023-04-13:
I successfullly launch Drastic by adding a shell script to `/userdata/roms/ports` folder and launch the script in ports section in Batocera main menu. However, there are still some problems.
```
cd /userdata/system/configs/drastic >> drastic.log
./drastic >> drastic.log
```
**Problem 1: If I run drastic without redirecting its output to a file like  `drastic.log`, drastic will crash just like running roms directly in Batocera main menu. Hope this information will be helpful.**
Problem 2: Configuration inside Drastic can not be saved.
Problem 3: There are actualy another approach to launch drastic. You can just open the file manager and click the drastic binary to start it. But this approach requires a otg cable with a keyboard and a mouse.

PS1: You need to at least run drastic once from Batocera main menu to generate files in `/userdata/system/configs/drastic`
PS2: I tried Pokemon HeartGold and it run at full speed. This is very impressive. But switching screens to play pokemon is very painful. Can someone recommend nds games which don't need switch screens frequently?

rth3, 2023-04-13:
Any chance you could provide a step-by-step on how to do the workaround?

rth3, 2023-04-13:
Also, Mario Kart DS and New Super Mario Bros do not require frequent screen switching.

acmeplus, 2023-04-14:
Here's a full script based on the one posted by @TsukiNoKnight that allows you to launch drastic standalone from the ports section. Just uncompress the zip file and copy drastic.sh to roms/ports/ folder.

[drastic.sh.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/11228145/drastic.sh.zip)

rogerillu, 2023-04-14:
I have tested it. It's working like a charm: new super mario bross, and mario kart for DS. Thanks for sharing!!!

rth3, 2023-04-14:
Seems like running drastic from ports section gives me a black screen. Any knowledge on this? Logs didn't offer any helpful information.

rth3, 2023-04-14:
> > Seems like running drastic from ports section gives me a black screen. Any knowledge on this? Logs didn't offer any helpful information.
> 
> I have copied a couple of roms in the nds folde. When you boot batocera, you will see the games in the nds. Go in, and push select button. Switch the emulator and choose drastic. Then try to load a game (will fail).
> 
> Once donde that, go to ports and launch the drastic.sh script (it need to be copied in the ports folder to appear there). If it does not work, try to reboot and launch drastic from ports again.

This worked perfectly. Thanks so much everyone!

facemasktuxedo, 2023-04-16:
Awesome, I did have to remap buttons, but worked like a charm on Phoenix Wright games

satencott, 2023-04-27:
Just wanted to let everyone know that as of 0.8 update, the emulator selection toggle in the NDS system settings is currently broken. This makes it impossible to use @TsukiNoKnight's method to generate the necessary config files to run Drastic from the ports folder. 

If you have a copy of your `/userdata/system/configs/drastic` folder backed up from a previous build, you can paste it into your 0.8 build and it should work allow Drastic to launch from ports again. 

I'd recommend for anyone looking to update to 0.8 and play NDS games to make a of backup their files before updating, so that they can just transfer them over for the time being.

acmeplus, 2023-04-28:
For all that have issues with drastic running as an standalone port:

Update ``roms/ports/drastic.sh`` as follows:
```
#!/bin/sh

if [ ! -d /userdata/system/configs/drastic ];then
	mkdir -p /userdata/system/configs/drastic
	cp -r /usr/share/drastic/* /userdata/system/configs/drastic/
fi

cd /userdata/system/configs/drastic >> /drastic.log
drastic >> drastic.log
```

Now refresh your game lists (Settings -> Game Settings -> Update Gamelists) and run drastic from the ports section. Enjoy!

kirbydoggs, 2023-04-28:
What do I do after running drastic port? @acmeplus

acmeplus, 2023-04-28:
> What do I do after running drastic port? @acmeplus

Drastic should be running and you can navigate to /userdata/roms/nds where your roms should be located. Note that drastic expects roms in .nds format

kirbydoggs, 2023-04-28:
I accidentally posted this comment elsewhere but I ran into another weird issue.. I can’t access the share partition on Windows anymore..

sankeriden, 2023-05-05:
> ¡Hola a todos! No sé si alguien más tendrá el mismo problema, pero para que funcionara también tuve que cambiar los permisos de lectura y escritura, el script no pudo crear drastic.log y el emulador simplemente no se inició. Lo hice para toda la división por si acaso `sudo chmod 777 ./ -R` Ahora funciona de maravilla <3

> ¡Hola a todos! No sé si alguien más tendrá el mismo problema, pero para que funcionara también tuve que cambiar los permisos de lectura y escritura, el script no pudo crear drastic.log y el emulador simplemente no se inició. Lo hice para toda la partición por si acaso `sudo chmod 777 ./ -R` Ahora funciona de maravilla <3

How did you do it? Can you Teach me

zopzop76, 2023-05-05:
I'm having the same issue.  The script won't write to the card.  The directory or the log file isn't created.  @vajra108 How did you manage to use the sudo chmod command to make it write to the sd card?

vajra108, 2023-05-06:
Hi!
I did in on a linux terminal, just put the card in my PC.. if you don't use
linux I'm sure there are alternatives.
Another person on discord installed batocera on a fat32 card instead of
flashing and they said it was working as well

Oatise, 2023-06-11:
I am having issues with the "drastic.log" file not appearing in the file location

Kaymations, 2023-06-12:
> Hi! I did in on a linux terminal, just put the card in my PC.. if you don't use linux I'm sure there are alternatives. Another person on discord installed batocera on a fat32 card instead of flashing and they said it was working as well
> […](#)

Could you please explain what you did exactly. I am not on linux but more info might help me find an alternative method

gingertastrophe, 2024-01-21:
> > What do I do after running drastic port? @acmeplus
> 
> Drastic should be running and you can navigate to /userdata/roms/nds where your roms should be located. Note that drastic expects roms in .nds format

How do you navigate through the menus in drastic though? The d-pad isn't inputting as expected. The up button brings me to the previous directory and the down button just brings me to the last item in the dir. No other buttons do anything. I think it's assuming I have KB input.