rg35xx-cfw/rg35xx-cfw.github.io #29: Dreamcast, laggy choppy sound

facemasktuxedo, 2023-04-16:
Using flycast and default settings, sound and gameplay is still choppy even with shaders off.

neocvera, 2023-04-16:
To be expected as this system is a bit above the devices pay grade. There may be specific settings and tweaks that can improve things but default settings - your experience is common.

eroldru, 2023-04-20:
> Using flycast and default settings, sound and gameplay is still choppy even with shaders off.

I have not tested it yet, but if not done yet, please check the DC bios is present. That makes a lot of difference if the case. Also keep in mind that the device is not made for DC and it is just a bonus. You could also try switching the core settings and try getting better performance. Also, where possible use PAL roms. They are capped at 50hz so 25/50 FPS which at some cases will make a difference.