rg35xx-cfw/rg35xx-cfw.github.io #31: v6 Cant get wifi to work at all

stapo101, 2023-04-16:
I have a RTL8192 usb wifi adapter and a USB A to USB C adapter.  I cant get wifi working at all.  Tried having it plugged in when booting, having it plugged in after booting, plugging in the USB A to USB C first etc.  Also tried manually inputting the SSID but nothing is working.  Any clues? Thanks

acmeplus, 2023-04-16:
Do you know which revision of the board you have? IF not, is it a fairly new model or an older one?

stapo101, 2023-04-16:
I've had it about 4 weeks since new, how can I tell?

stapo101, 2023-04-16:
this is the adapter I am using

stapo101, 2023-04-16:
https://www.amazon.co.uk/dp/B08HYM4829?psc=1&ref=ppx_yo2ov_dt_b_product_details

acmeplus, 2023-04-16:
That seems to be a fairly new 8192 adapter. The one supported by the board is a 8192cu that is not 802.11n but 802.11b. It does support 802.11n with a TP-Link 8188eu adapter.

Try to use a different usb device to see if it's detected (e.g. a usb keyboard/mouse). If those are detected then the issues with the usb wifi chipset not being compatible.

stapo101, 2023-04-17:
Ok. Latest update...I have tried a USB mouse and that doesn't work at all either. If that means it's a board issue is there any possibility of a future fix? I'm happy to test it.

acmeplus, 2023-04-17:
Yes, most likely will be fixed, there was an older version that had usb working on most revisions, but it didn't have all the updates required for other boards to work.

acmeplus, 2023-04-18:
> Ok. Latest update...I have tried a USB mouse and that doesn't work at all either. If that means it's a board issue is there any possibility of a future fix? I'm happy to test it.

Try release [v0.4](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/releases/download/rg35xx_batocera_lite_alpha_v0.4/batocera_lite_rg35xx_20230410.zip), it may work on your board. Do the usb keyboard/mouse test too and let me know if wireless now works.

stapo101, 2023-04-18:
Ok, I;ve tried v4.  Still nothing I'm afraid.  No mouse, no keyboard, now wifi.  I have double checked that the USB c to a adapter is working on another device and it is.

acmeplus, 2023-04-22:
> Ok, I;ve tried v4. Still nothing I'm afraid. No mouse, no keyboard, now wifi. I have double checked that the USB c to a adapter is working on another device and it is.

Try the newest 0.7 version. I've heard some v6.0 users indicating it's now working.

stapo101, 2023-04-22:
Thanks.  Just tested 0.7 with mouse, keyboard and wifi adapter.  Still not working I'm afraid.

acmeplus, 2023-04-28:
@stapo101 some users with a similar issue have indicated that wireless works for them after the console has been off for a while. In those cases after the console has been off for 20-30 min the wireless is detected. Not a solution, but may be something to check.

luci-11, 2023-04-30:
@stapo101 did you try with version 0.8?  USB for me is definitely more stable. it is always detected and wifi working with no issues with Edimax adapter

stapo101, 2023-04-30:
> 

Just tried latest release 0.8. still nothing I'm afraid.  It sure USB is working at all other than charging.

ChrisGetHub, 2023-05-05:
I have the same issue. I am using the TP-Link TL-WN725N Nano USB WLAN Dongle and it doesn’t work.  I suspect the USB port simply doesn't recognise anything on hardware

demichel93, 2023-06-03:
Same problem here, I have batocera 0.8.1m, I got TP-Link WN725N which I use with Playstation Classic with Autobleem or whatever the OS available is there. I use USB-C to A OTG adapter, Batocera on RG35XX does not see any networks, I played around with uImage files, switching adb to uImage and reverse, nothing seems to work. USB-C to A adapter does work on other devices.