rg35xx-cfw/rg35xx-cfw.github.io #34: 0.7 version issues

rogerillu, 2023-04-21:
- when turning on, it starts but then there's a black screen, and you need to press the power button again to complete the turn on

- nintendo ds script does not work anymore and if you try to start the game from batocera menu, the emulators are asking for bios, but do not start.

- performance seems to be slow

- language change is still not working

Besides that, great improvements, and very good job. Is insane we can play n64 or nds in this handled.

acmeplus, 2023-04-22:
> * when turning on, it starts but then there's a black screen, and you need to press the power button again to complete the turn on

Yes, it's part of the initial implementation of the charger. Will be corrected once the implementation is optimized.

> * nintendo ds script does not work anymore and if you try to start the game from batocera menu, the emulators are asking for bios, but do not start.

Will check.

> * performance seems to be slow

Try to send some specific information, a specific system/rom and a before/after performance measurement. Also make sure you are using the same OC settings.

> * language change is still not working

Yes, not corrected, see #32 

> Besides that, great improvements, and very good job. Is insane we can play n64 or nds in this handled.

Yes! no one expected that :)

TsukiNoKnight, 2023-04-22:
The drastic script works fine for me. But ds roms won't launch directly with any core.

acmeplus, 2023-04-22:
> The drastic script works fine for me. But ds roms won't launch directly with any core.

I can confirm the standalone roms are broken. I thought they were fixed but the last recompile broke them. Will update (for good this time) in the next release

Cristovao-Sousa, 2023-04-27:
with my experience i also so find performance to be low.
screen tear in all emulators that i tried.
Sega MD-Picodrive/GenesisPlus GX - tested for example playing Sonic 1.. moving the character left and right.. very noticeble.
SegaSMS-Picodrive/GenesisPlus GX - tested for example playing Sonic 1.. moving the character left and right.. very noticeble.
FBNEO - FBalpha/FBNeo - Cadillacs and Dinosaurs - intro and moving the character through the level also very noticeable.
SNES - Snes9X - Super Mario All Stars - scrolling through the games or playing any of the games... also very noticeable.

Also turn off bilinear filtering... overclocked .. but didn't get better results.

acmeplus, 2023-04-28:
> with my experience i also find performance to be low. screen tear in all emulators that i tried. Sega MD-Picodrive/GenesisPlus GX - tested for example playing Sonic 1.. moving the character left and right.. very noticeble. SegaSMS-Picodrive/GenesisPlus GX - tested for example playing Sonic 1.. moving the character left and right.. very noticeble. FBNEO - FBalpha/FBNeo - Cadillacs and Dinosaurs - intro and moving the character through the level also very noticeable. SNES - Snes9X - Super Mario All Stars - scrolling through the games or playing any of the games... also very noticeable.
> 
> Also turn off bilinear filtering... overclocked .. but didn't get better results.

Please give a try to v0.8, it should fix those issues.

acmeplus, 2023-06-07:
Closing due to inactivity.

closed 2023-06-07 by acmeplus