rg35xx-cfw/rg35xx-cfw.github.io #35: PS1 (PSX) games are having shuttering and choppy sound with laggy and low FPS  (Tested on 0.5 -> 0.7)

nmqanh, 2023-04-21:
Hello, 

First of all, thank you so much for your great effort in developing Batocera for RG35xx. Look forward to its future.

I have been testing PS1 (PSX) games with default settings, but the performance is pretty bad compared to GarlicOS. Here are my current settings
- Default Settings for all
- Verified that Enhanced Resolution was set to OFF
- Overclocked to EXTREME / TURBO
- I am on PCB V6 that was shipped with 2100mah Battery. 
- Full Bios pack were installed for PS (There is no missing bios)
- Use recommended CHD roms for PSX
- Even tried with Frame Skipping settings but no luck.

I have tried the following games.
- Need for Speed III: Hot pursuit: the sounds from the introduction video + sound in menu is choppy. When racing, FPS is around 30fps and sound is still choppy (however, it's more choppy in menu)
- Final Fantasy IX: sounds are shuttering / choppy videos of game, in battle is very laggy and hard to play. FPS is around 15-20.
- Harry Potter: Chamber of secrets: sounds are shuttering / choppy, FPS is low.

I have also been playing these games on GarlicOS, and they can easily reach FPS 60 with no choppy sound on default settings (Enhanced Resolution OFF). On GarlicOS, when setting Enhanced Resolution to ON, it even gets FPS 60 for some games e.g. FInal Fantasy IX 

Thank you so much for reading the issue and look forward to more testing releases.

P/S: On Batocara CFW 0.7, When I disable sound, the performance is very fast even with Enhanced Resolution ON.

XQuader, 2023-04-21:
I have clear Batocera 0.7 (just flashed), and use the same Roms SD card as Garlic OS (they share the same Roms and Bios folders).
Performance in Gran Turismo 2 (Turbo Overclock, Enhanced Resolution + Speedhack, PS CPU Clock 80) is the same as in Garlic OS with overclock++ (same settings).

nmqanh, 2023-04-22:
Thank you @XQuader , I have just tried to install with a clean 0.7 (flash rather than the copy and paste method) and can confirm that the performance is at least the same or even better than GarlicOS. Enhanced Resolution with Speedhack is still producing choppy sound with somewhat laggy though on Harry Potter games and Need for Speed III: Hot pursuit.

closed 2023-04-22 by nmqanh