rg35xx-cfw/rg35xx-cfw.github.io #42: Emulator performance in v0.8 is really sluggish

evolver79, 2023-04-28:
See attached videos for NES and GBA performance
Was fine on 0.7. Did a fresh install writing new image to sdcard.

Tested gb, gba, nes, snes, pcengine and megadrive. All the same issue.
Funny thing is that dreamcast and psp runs better.

https://user-images.githubusercontent.com/21036181/235128687-d5a01333-384e-431c-8ed3-d825de90f8dc.mp4
https://user-images.githubusercontent.com/21036181/235128657-6cb1c475-de4a-42e0-a7ba-b65a00c6fae9.mp4

acmeplus, 2023-04-28:
Thanks for the report. Yeah, I suspect the issue only impacts retroarch. You can try to change the batocera.conf variable ``global.retroarch.audio_driver=sdl2`` to alsa

evolver79, 2023-04-28:
@acmeplus I'll give it a try later today and report back :)

acmeplus, 2023-04-28:
Just an fyi. I've  just tried adventure island and I don't the same issue you get in your video.

evolver79, 2023-04-28:
Aha, I have another theory aswell... i'm using sd slot 2 for roms created when running v0.7, so might have a config from 0.7 in there making a mess/faulty config.

acmeplus, 2023-04-28:
> Aha, I have another theory aswell... i'm using sd slot 2 for roms created when running v0.7, so might have a config from 0.7 in there making a mess/faulty config.

Definitely more likely. You need to copy the new variables in ``batocera.conf`` to your sdcard ``system/batocera.conf``

evolver79, 2023-04-28:
That was it, the config needed to be replaced on the second sd card. You can close this.  Thanks for the help! :)

closed 2023-04-28 by acmeplus