rg35xx-cfw/rg35xx-cfw.github.io #45: Performance regression on some cores/emulators from v0.7 to v0.8+

acmeplus, 2023-05-05:
Several users report a performance degradation after upgrading to v0.8/v0.8.1. 

In order to investigate this issue I need specific examples of roms/cores and which metrics to test (FPS, general responsiveness, etc.), specific settings for the emulators (frameskip, video threaded, etc.).

Please add examples and metrics in the comments below.

ANhan88, 2023-05-05:
J2ME games on Rg35xx ?

MauricioSuporte, 2023-05-05:
I Played Contra 3 Snes on the default core, and I noticed some slowdowns during gameplay that don't happened in 0.7

cyberka1, 2023-05-08:
Psx perfomance was bad on v0.8 but now on 0.8.1 is even worst all set in AUTO in all systems. new flashed image.

acmeplus, 2023-05-11:
There's an issue with the new audio routing that is impacting performance. There will be potentially a fix in the next release, but it may have some issues (volume going back to 100% when launching a rom).