rg35xx-cfw/rg35xx-cfw.github.io #48: BIOS not recognized

mvergaraq, 2023-05-09:
Hi everyone! I'm having a hard time making the bios files work, I get the "missing bios file" message everytime

I placed them into the root "bios" folder of the sdcard that batocera (0.8.1) creates but it won't read any

I also tried placing them inside roms/bios, as this is the directory written in the release instructions but it won't work either

I also tried setting the bios manually but if no NDS game boots then I can't access retroarch settings, I tried launching a gba rom and changing the settings from there but it won't save the directory (even after going and saving the configuration file)

mvergaraq, 2023-05-09:
Guys, so sorry! I wasn't used to the bios naming conventions of batocera, no missing bios error now but the games just won't boot, black screen and back to game select (nds)

zopzop76, 2023-05-09:
> Guys, so sorry! I wasn't used to the bios naming conventions of batocera, no missing bios error now but the games just won't boot, black screen and back to game select (nds)

Yup.  DS emulation is currently not working in Batocera for the RG35xx.  There is a workaround to try and get it working here : https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/issues/25

Works for some people but not for others (like me).  Hope it works for you.