rg35xx-cfw/rg35xx-cfw.github.io #49: PSP experience with version 0.8.1

AskarTuraev, 2023-05-10:
On the PSP control settings on the up-down-right-left buttons uses the cross, respectively, I can not put blank values on these buttons, and I could not set the movement due to the fact that the display of buttons on the PSP is not visible - incorrect display of the PSP on the screen. Having applied the settings once, I could not reset them, and because of the same appointments, the settings menu could no longer be used. Also would like some sort of guide to setting up psp and n64 emulators or preconfiguration in future versions of batocera. 
Can you contact me at: **aturaev42@gmail.com**

acmeplus, 2023-05-11:
Settings for those emulators are typically saved on `system/configs/ppsspp` or in `system/batocera.conf`. Deleting the whole system/configs/ppsspp should fix that issue.

For N64 controls you can check this reddit post that includes some information: https://www.reddit.com/r/RG35XX/comments/13dpert/configuring_nintendo_64_controls_batocera_08_or/?utm_source=share&utm_medium=ios_app&utm_name=ioscss&utm_content=1&utm_term=14

AskarTuraev, 2023-05-20:
hello again, ty, сould you help with the pictures to the games, how many
tried png files do not work, added -image, also failed, is there a manual?