rg35xx-cfw/rg35xx-cfw.github.io #50: Random button presses on

Schaefers, 2023-05-19:
I'm seeing random button presses on `37-dev`, my mainboard is `RG35XX_V6C 2023-01-30`. I'm using batocera on a SandDisk memory card. They're not occuring with the original OS on the bundled sd card. 

[image]

Schaefers, 2023-06-04:
Apparently this has been a broken unit, got another one with the same mainboard and don't have random button presses there. Closing here.

closed 2023-06-04 by Schaefers

acmeplus, 2023-06-04:
> Apparently this has been a broken unit, got another one with the same mainboard and don't have random button presses there. Closing here.

Glad that it was resolved. There are some hardware variations even with the same revisions that make it very hard to determine what’s hardware or firmware related.