rg35xx-cfw/rg35xx-cfw.github.io #51: menu button not working

dooeif5, 2023-05-29:
all buttons work except the menu so i cant save or exit games

acmeplus, 2023-06-07:
Try to use an external keyboard and remap the controls in batocera. Alternatively you can delete the file system/configs/emulationstation/es_input.cfg and try again.

descalzi, 2024-01-18:
Same, I tried deleting "es_input.cfg" using the app in Ports. Once I exit the app it automatically goes into the keys-mapping, which rg35xx-plus doesnt have analog-arrows .. so I rebooted but menu key still doesnt do anything

acmeplus, 2024-01-18:
> Same, I tried deleting "es_input.cfg" using the app in Ports. Once I exit the app it automatically goes into the keys-mapping, which rg35xx-plus doesnt have analog-arrows .. so I rebooted but menu key still doesnt do anything

This is a very old ticket for the rg35xx during the development. It does not apply to the current firmware for the RG35XX or for the RG35XX plus.

What's the issue that you have? Note that in EmulationStation the Menu button is not mapped to menu but instead used as a hotkey. For instance Menu + START exits the emulator.

closed 2024-01-18 by acmeplus