rg35xx-cfw/rg35xx-cfw.github.io #56: System does not recognize USB Controller

daywedie, 2023-07-18:
I'm trying to use external usb controllers with Batocera on RG35XX, but it doesn't seem to recognize any of them in the system input settings. How would I go about to make it work? I've tried with a dualshock 4 and a generic one, no luck with both.

daywedie, 2023-07-18:
By the way, I'm trying this with n64 games and don't seem to be able to get into retroarch's menu with them.

tabletramos, 2023-12-01:
Yesterday I were all day trying several things with: "Dreamcast", "Psp", "N64", "Nds", "Psx".

and I saw that the USB connection was not working. I have a generic joystick with an "usb - usb C" adaptor that works with "Garlic OS" and the "Native OS".

It would be amazing if you could fix this soon

TomChino, 2024-01-18:
Using koriki, my USB controller only works when I launch Emulation Station.
Using simplemenu, it's not recognized.

acmeplus, 2024-01-18:
Yes, that’s correct. Simplemenu does not support external controllers. You can map the controller to work on Retroarch though.