rg35xx-cfw/rg35xx-cfw.github.io #57: Porting m8c tracker USB problem.

thirdegg, 2023-09-06:
Hello! I'm trying to port m8c (https://github.com/laamaa/m8c). Using the provided set of tools, I managed to compile the project with libusb dependencies. The application starts but doesn't detect the Teensy 4.1 connected via OTG. In the application logs, it says libusb_open_device_with_vid_pid == null. I've tried various ways to initialize the device, but it seems like I lack experience in this matter. I've tried usbdbg in host mode and vice versa. I also attempted to capture events in udev, but it's quiet there. I would appreciate it if you could point me in the right direction.

acmeplus, 2023-09-07:
Very nice! Would love to have that added to the ports.

About usb, note that you cannot use the ADB (adb_uImage) kernel and have access to usb at the same time. You have to use the regular kernel (uImage) that has support for the USB. This is an inconvenience with this kernel that I've not yet figured how to to solve to change that dynamically.

That being said you can enable terminal on the rg35xx and connect a powered usb hub with a keyboard so you can use it there. Once you are in the system you can use regular linux commands like lsusb and dimes to list usb devices and check for the latest changes and see if plugin in the Teensy triggers and message (hint you should be able to plug a regular monitor and access the console in a larger screen).

I don't have much time these days, but I do have a couple of Teensy's around, so let me know after you run those tests. 

Let me know if you need more help.

thirdegg, 2023-09-11:
Thank you for your response! It seems there are issues with USB device initialization. I've tried connecting various devices like a flash drive, keyboard, mouse, and joystick via OTG without ADB, but none of them initialize. There don't seem to be any issues with the OTG converter, as I've tested it on my Android smartphone, and the USB port on the console itself is fine since ADB works. I've also tried these actions on GarlicOS, where the joystick and keyboard initialize and work. There are even attempts to communicate with the USB flash drive (the LED indicator blinks). I have limited experience with embedded systems, and it feels like I've exhausted all available methods. Do you have any suggestions on what the issue might be? Are there any system configuration parameters that might help? Thanks in advance!

hihihihinick, 2024-01-27:
I’m taking the weekend to figure this out. I purchased an RG35XX plus and best I can tell by looking at the H700 cpu data sheet as well as boppin’ around, looking at kernel modules and device overlays for other hardware using the same cpu it’s a amendable issue with a custom kernel compilation.