rg35xx-cfw/rg35xx-cfw.github.io #58: No cursor-control in scummvm

mr-who2012, 2023-09-17:
In scummvm, I can't control the cursor with the d-pad. Therefore it's impossible to navigate the scummvm-menu, the on-screen-keyboard or play any game. All other buttons work fine, so I can toggle keyboard and menu without problems. But without cursor it's not even possible to quit scummvm. It doesn't matter which scummvm I use, the libretro-one or the scummvm-native.

acmeplus, 2023-09-17:
Yes, it's a known bug for the standalone scummvm. It's not clear why scummvm is ignoring the control changes in the configuration. 

Note that the libretro version works correctly.

mr-who2012, 2023-09-17:
Ok, I will look into libretro-version (and it's configs) again. Meanwhile: Is it possible to actually replace the standalone-version somehow? I did read somewhere that the newest scummvm may hav issues, older versions don't. The scummvm-version in batocera is newer than the official version downloadable on the homepage!

mr-who2012, 2023-09-17:
scummvm-libretro is crashing after intro of the dig and fate of atlantis. When I pres Start-Button before, I am getting the menu but cant move the cursor.
I have to mention that I converted all filenames of both games to lowercase, since batocera is using an ext4-filesystem on rg35xx and DOS-games (and FAT) don't know such things like "case-sensitivity", as ext4 does. 
In each game-folders is a file dig.scummvm and atlantis.scummvm, respectively.

acmeplus, 2023-09-18:
The only way to change or add a new scummvm is to add it to the sdcard in a specific path (e.g. bin/scummvm) and then modify the system/.simplemenu/launchers/scummvm_sa to point to the new bin/scummvm. The path of the SDCARD is /userdata, so if you initially had /usr/bin/scummvm you need to replace that with /userdata/bin/scummvm.

Note that this assume you have a compatible scummvm binary or know how to compile it from scratch.

mr-who2012, 2023-09-18:
> Note that this assume you have a compatible scummvm binary or know how to compile it from scratch.
yep, this is indeed a problem. I don't know where to get the proper arm7-version for rg35xx. There ARE a bunch of exotic versions on the[ scummvm-download-page](https://www.scummvm.org/de/downloads/) but I don't know which was compatible (and bug-free). 

changing files on the sd-card shouldn't be too tricky. Usually the files are stored within container-files which need to get mounted with loop-devices in linux. It's usually not as difficult as it sounds. Not sure if I need to modify config-files when replacing the binary. Overwriting the original should preserve all paths. 

But no...I don't wanna compile scummvm from source myself...what if I copied scummvm from garlicos...somehow? I know their version is running fine.

acmeplus, 2023-09-18:
You will have to wait until I have time to update it then. The garlic version is not compatible because it uses a different architecture.

About changes, yes, you can just open the rootfs (batocera) mount it on your pc, make changes and squashfs it again.

mr-who2012, 2023-09-19:
oh, alright then. Thanks for the info...and your intention updating scummvm! I didn't realize I was directly talking to the proper dev! 

So...in this case...it might be offtopic...but I see it as a rare chance: Is there any support..forums...wikis or alike for running amiga-games on batocera? Not a single game is running. On garlic, most of them, in my collection, did. And yes, I installed and double-checked all bioses and roms. 
Not being able to run scummvm and amiga is a real shame, since batocera is actually so much better than the buggy garlicos! Amiga-games are generally difficult to set-up, even on PC. Being able to run them (almost) hassle-free on RG35XX is a big reason using that device.

acmeplus, 2023-09-19:
> oh, alright then. Thanks for the info...and your intention updating scummvm! I didn't realize I was directly talking to the proper dev!
>
>
>
> So...in this case...it might be offtopic...but I see it as [...]

puae2021 runs really well. Obviously understanding you are not going to get a 1:1 performance compared with a real Amiga but good enough for this device. Check the type of rom files you have and also change the core you use and select puae2021. Some cores accept adf files and others WHDload. Neither amiberry or uae4arm are working well at this point.

mr-who2012, 2023-09-19:
Thank you. I will try that.

mr-who2012, 2023-09-21:
No, can't get Amiga to work. Both, Amiga and AmigaCD32 don't start any games with any core. Using PUAE2021 even crashes Batocera. SNES-MSU-1 is working wondeful ootb, btw. Oh...and broken headphone-detection...is super-annoying...though, not as terrible as the broken charger, forcing the device to be turned on for hours, playing music with illuminated display.

1alien1, 2024-04-08:
Any news on getting Scummvm to work with controls?