rg35xx-cfw/rg35xx-cfw.github.io #65: nintendo DS not bios Found

H0unterx, 2023-12-06:
I try many bios and none is working. Is a bug?

acmeplus, 2024-01-17:
The bug is not the bios, drastic includes the required files already. Check the latest release and report back (https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/releases/tag/rg35xx_plus_batocera_lite_koriki_alpha_20240116)

gingertastrophe, 2024-01-21:
I have the same issue selected DS games from EmulationStation. I don't see anything that resembles "drastic" in the ports area, is this something I have to add myself, even on the latest release?

james-mccloud, 2024-01-31:
I have the same problem with 20240126 update. First time the message “not BIOS found” appeard i updated bios folder but now the problems is the system does not recognize any emulator, the game does not star and return to game select menu.

thank you 👍🏻

james-mccloud, 2024-01-31:
Ok I think the problem is solved. I was launching the macOS metadata archive. Do you know any way to ignore those archives ?

acmeplus, 2024-01-31:
It's always a pain with the Mac, you have manually erase those ._ files.... using a shell script or something similar.

james-mccloud, 2024-01-31:
Ok, I need help for button mapping configuration for drastic. I do not find  “es_systems_nds.cfg” file. 

Thank you.