rg35xx-cfw/rg35xx-cfw.github.io #70: rg35xx plus doesn't really turn off ?

vvzpaco, 2024-01-18:
Even after turning off the console by selecting shutdown through the batocera menu, it completely turns off the screen, but continues to heat up, I did this process at night and went to sleep, when I woke up and went to turn it on, the battery was completely empty.

acmeplus, 2024-01-18:
Ok, I'll investigate, I'm running my tests with the device open and without battery, so that may be part of the reason I didn't notice that before.

korikirg35xx, 2024-01-18:
The same happens to me on my original rg35xx

docloulou, 2024-01-18:
Yes, same for me this morning. 
Wondering if a PORTS script with `shutdown ` command is a valid temporary fix ?

acmeplus, 2024-01-18:
> Yes, same for me this morning. Wondering if a PORTS script with `shutdown ` command is a valid temporary fix ?

That's definitely an option as a temporary, you can call the ``poweroff.sh`` script from a shutdown script in ports. This should do:
```bash
#!/bin/sh

poweroff.sh
```

docloulou, 2024-01-18:
Thank you @acmeplus , for your response. However, I am curious about the difference in impact between calling this script via PORTS and via the UI (shutdown with saving metadata).

rupokify, 2024-04-05:
> > Yes, same for me this morning. Wondering if a PORTS script with `shutdown ` command is a valid temporary fix ?
> 
> That's definitely an option as a temporary, you can call the `poweroff.sh` script from a shutdown script in ports. This should do:
> 
> ```shell
> #!/bin/sh
> 
> poweroff.sh
> ```
> 
> Just save that in roms/ports as `Shutdown.sh`

For me, it doesn't work because while it tries to gracefully shut down by stopping the process, the NFS daemon fails to stop. So the device doesn't shut down, but rather restarts back and the screen keeps flickering (not sure why) until I go to the menu.

[image]

Working alternative: 

```
#!/bin/sh
shutdown -h now
```