rg35xx-cfw/rg35xx-cfw.github.io #71: Cannot connect to wifi with Plus

ViBE-HU, 2024-01-18:
I did  fresh install with the latest build and somehow even if it see the wifi networks it does not connects. The stock Anbernic OS can connect without any issues.

I have an Asus router and I only find these in the log:
```
Jan 18 16:07:00 wlceventd: wlceventd_proc_event(685): eth7: Auth 4A:xx:xx:xx:xx:92, status: Successful (0), rssi:0
Jan 18 16:07:00 wlceventd: wlceventd_proc_event(722): eth7: Assoc 4A:xx:xx:xx:xx:92, status: Successful (0), rssi:-47
Jan 18 16:07:00 hostapd: eth7: STA 4A:xx:xx:xx:xx:92 IEEE 802.11: associated
Jan 18 16:07:00 hostapd: eth7: STA 4A:xx:xx:xx:xx:92 RADIUS: starting accounting session AB1D2A903A87EBC9
Jan 18 16:07:00 hostapd: eth7: STA 4A:xx:xx:xx:xx:92 WPA: pairwise key handshake completed (RSN)
Jan 18 16:07:01 wlceventd: wlceventd_proc_event(645): eth7: Deauth_ind 4A:xx:xx:xx:xx:92, status: 0, reason: Station requesting (re)association is not authenticated with responding station (9), rssi:0
Jan 18 16:07:01 wlceventd: wlceventd_proc_event(662): eth7: Disassoc 4A:xx:xx:xx:xx:92, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Jan 18 16:07:01 hostapd: eth7: STA 4A:xx:xx:xx:xx:92 IEEE 802.11: disassociated
Jan 18 16:07:04 wlceventd: wlceventd_proc_event(685): eth7: Auth 48:xx:xx:xx:xx:92, status: Successful (0), rssi:0
Jan 18 16:07:04 wlceventd: wlceventd_proc_event(722): eth7: Assoc 48:xx:xx:xx:xx:92, status: Successful (0), rssi:-50
Jan 18 16:07:04 hostapd: eth7: STA 48:xx:xx:xx:xx:92 IEEE 802.11: associated
Jan 18 16:07:04 hostapd: eth7: STA 48:xx:xx:xx:xx:92 RADIUS: starting accounting session 137C233ACB067ABE
Jan 18 16:07:04 hostapd: eth7: STA 48:xx:xx:xx:xx:92 WPA: pairwise key handshake completed (RSN)
Jan 18 16:07:05 wlceventd: wlceventd_proc_event(645): eth7: Deauth_ind 48:xx:xx:xx:xx:92, status: 0, reason: Station requesting (re)association is not authenticated with responding station (9), rssi:0
Jan 18 16:07:05 wlceventd: wlceventd_proc_event(662): eth7: Disassoc 48:xx:xx:xx:xx:92, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Jan 18 16:07:05 hostapd: eth7: STA 48:xx:xx:xx:xx:92 IEEE 802.11: disassociated
```

TrilIiard, 2024-01-21:
I found out that the problem is wpa3 or wpa2+wpa3 at the same time. For now you can only use wpa2 and 2.4Ghz for WiFi to work

ViBE-HU, 2024-01-21:
> I found out that the problem is wpa3 or wpa2+wpa3 at the same time. For now you can only use wpa2 and 2.4Ghz for WiFi to work

You are right. Since every network devices using WPA3 mixed I did not notice any difference. Now I set one router to WPA2 only and the console connected.

francoislg, 2024-02-09:
Currently using WPA3+WPA2 with `20240208`, I can see the correct SSID, double checked the password, so I guess the following

> Added potential fix for compatibility with WPA3 wifi routers

didn't work 😓

tonyshackelton, 2024-02-09:
I also run WPA2+3 and the newer build with the correction still doesn't connect for me either, WPA2 only works as does the stock firmware. Keep up the good work!

ViBE-HU, 2024-02-10:
I just checked. Yeah, it is still an issue.

Venlaw, 2024-02-11:
Disabling WPA3/2 mode along with Protected Management Frames (PMF) worked for myself. Both 5 and 2.4Ghz works.

acmeplus, 2024-02-11:
> Disabling WPA3/2 mode along with Protected Management Frames (PMF) worked for myself. Both 5 and 2.4Ghz works.

I’m curious if just disabling PMF would work. I’ve tested with a simple portable router with wpa2/wpa3 and couldn’t reproduce the problem. But it didn’t have PMF so just thinking that may be part of the issue

ViBE-HU, 2024-02-11:
@acmeplus PMF as i understand it is a requirement of WPA3.
https://www.wi-fi.org/knowledge-center/faq/what-are-protected-management-frames

and it is not even possible to disable it on my router.

tonyshackelton, 2024-02-12:
I run a 2nd ssid that's just set for WPA2 for legacy devices, same AP(unifi) as my main WPA3 network. It connects to the WPA2 ssid without problems, WPA3 does not connect. I don't think I can disable PMF specifically, but I'm happy to test otherwise.