rg35xx-cfw/rg35xx-cfw.github.io #72: RG35XX, L1 Button not working

MahirC3, 2024-01-18:
My L1 trigger doesn’t work at all. I’ve tested it using the IO setting as well as in multiple games. All other buttons function correctly. I have 0 clue how to fix this or if this is even fixable. Appreciate any help, thanks.

descalzi, 2024-01-23:
if it fails in the IO setting might just be your button is broken.
Does it work with the StockOS ? Doesnt sound like a Batocera specific issue ...

MahirC3, 2024-01-23:
> if it fails in the IO setting might just be your button is broken. Does it work with the StockOS ? Doesnt sound like a Batocera specific issue ...

Nope, doesn’t work at all. Every other button works so I assumed it was a hardware fault. Anyways, contacted the seller and I got a refund and got to keep it. Can’t be fussed at all.

catprisbrey, 2024-06-19:
I just bought an RG35xx and hit this similar issue. The L1 works in the IO test screen, but in some PS games like Metal Gear for example, L1 doesn't work, and also L2 is Turbo even after changing the setting to another button.

It works in other games/emulators, so far it's been primarily Metal Gear on PS where I've hit the issue.

catprisbrey, 2024-06-26:
Flashes to the lastest firmware, and in the setup steps changed the turbo assignment. Issues have gone away, everything is good now