rg35xx-cfw/rg35xx-cfw.github.io #76: rg35xx plus - Amiga games fail to show screen

ma7tcsp, 2024-01-22:
All kickstarts/roms found ok. 
game appears to load fine in the background as the music starts. The screen (time in bottom right, wifi icon in top right, controller icon top left) all flicker. 
Menu + B = takes me to Amiberry setup screen, but can never get back to the game. 
Tried many different screen settings/resolutions but never works.

acmeplus, 2024-01-26:
Yeah, it's an issue with amiberry. Try to change the emulator to use puae2021 or uae4arm.

FX50pro, 2024-02-11:
Hi guys,  Hope to get some help.
I see you maybe have the Amiga up and running in Koriki? However I'm really struggling with this one. Koriki throws me the MISSING BIOS FILES error....
I believe I have my RG35XX PLUS and Koriki set up correctly. I'm running the latest Update (fresh install) 20240208 on Card 1 and roms on card 2.
I downloaded the Mini cake 5.20 to V38 bios pack  (Internet archive) I tried others but no luck same Errors.
Would love to know yours works fine and pls point me in the correct direction of correct bios.
Thx in advance.