rg35xx-cfw/rg35xx-cfw.github.io #78: Videos not playing audio

VulcansMeMoRy, 2024-01-24:
I loaded a video onto sd card, and was able to access it via the "library" menu. The video plays well, but there is no audio. 

Probably not your priority right now, but having the ability to watch a show or movie on it would be awesome. 

Separate question... any plans for a browser in the future??? ;)

javonte-102, 2024-01-28:
I second this

VulcansMeMoRy, 2024-01-28:
I've tried everything I can think of. Something I have noticed, when using retrobat and also batocera on pc, you can toggle "video music" in the settings, when playing the preview of the highlighted game you are selecting, and it does indeed toggle on and off in retrobat and batocera for pc. The same toggle option for rg35xx plus batocera does nothing. All preview videos are silent aswel. I wonder if there is an underlying code issue which affects the game preview audio aswel as added movie files?

Secondary comment. Going back to watching videos on the rg35xx plus. I also noticed there were no "in video" controls like pause, rewind etc... so it was either watch it from the start or not watch it. In silence. Lol...

I appreciate all the hard work that goes into this platform, genuinely love it. Hope our comments here can help to polish it into a winning format :)

YooPita, 2024-02-09:
> All preview videos are silent aswel

This is true, I uploaded my videos using Skraper with the RECALBOX preset. I activated the sound for video in the sound settings and generally did everything as indicated [here](https://wiki.batocera.org/audio_issues#i_have_sound_in_emulators_and_the_main_menu_but_not_during_video_previews). Nothing helped and there is still no sound in the video snaps.

I think this is a fairly serious problem, since the sound in the menu is an important part of the functionality for me.