rg35xx-cfw/rg35xx-cfw.github.io #81: Battery indicator in EmulationStation

boolemancer, 2024-01-26:
I noticed that there's no battery indicator in EmulationStation on the RG35XX+/H and I decided to spend some time digging in to try to understand what's going on, and figured I'd share it here in case it's helpful.

It looks like ES is trying to read the battery information in `Platform::queryBatteryInformation` [here](https://github.com/batocera-linux/batocera-emulationstation/blob/df059dfe611e36dc0009c03ccef888d22adbf2b2/es-core/src/utils/Platform.cpp#L405C1-L406C30).

As you can see from the link, it's trying to find a device matching something like `/sys/class/power_supply/BAT0` that you might expect to find on a linux device, but it seems that the battery device on the RG35XX+/H is actually here instead:

```
/sys/class/power_supply/axp2202-battery
```

Which doesn't match the pattern they look for.

Curiously, the battery indicator does show up in RetroArch, and it looks like they try to find any file with `battery` in it in addition to the normal `/BAT#` files, so maybe it'd be simple enough to patch EmulationStation to do the same thing.

Unfortunately, I'm not very familiar with the world of buildroot and building custom firmware, so I haven't tried to make the changes myself yet, but maybe sharing what I found here will help.

acmeplus, 2024-01-26:
Thanks for spending time investigating this. Yes, there's a mismatch in the way the battery path is done for other regular linux devices that seems to be the problem. As some other users have pointed out retroarch has not issues reading the battery, and this is just an issue with the way ES and batocera parses the battery reading. The issue actually happens with both ES and the Batocera-info script.

acmeplus, 2024-01-26:
Actually, that pointed me to the right path, so it's now fixed.

[image]

Will be in the next update.

closed 2024-01-27 by boolemancer