rg35xx-cfw/rg35xx-cfw.github.io #83: no menu in dratic ?

vvzpaco, 2024-01-27:
In the new update, the menu button no longer opens the drastic menu, making it impossible to use save states and make other settings

acmeplus, 2024-01-27:
Which version 20240125 or 20240126? I'll check

vvzpaco, 2024-01-27:
20240125, I just saw that the 20240126 came out and I'm going to test it now

vvzpaco, 2024-01-27:
in 20240126 too, and the buttons seem to be a little disconfigured too D;

javonte-102, 2024-01-28:
Same issue here to all buttons are not mapped and it's no hotkey button to go into retro arch on 20240126

james-mccloud, 2024-02-01:
Me too, I do not find “es_systems_nds.cfg“ that seems to solve the problem

james-mccloud, 2024-02-01:
Ok I’ll do it and share what happen.

thanks

vvzpaco, 2024-02-01:
Try it and tell if it worked please, I couldn't 😔

james-mccloud, 2024-02-01:
Is the file mandatory to be installed on TF1 or it could be installed on TF2 ?

james-mccloud, 2024-02-01:
At this moment I have installed es_systems_nds.cfg and following the same folder route but in TF2 and still don’t work. 
I need to qualify that I’m playing on a RG35XX H. 
Hope this is helpful and thanks @acmeplus for his work.

vvzpaco, 2024-02-01:
I also tried on a spare SD card, but without success, i'm playing on a RG35XX PLUS and using version 20240122 for now 😔

acmeplus, 2024-02-01:
I’ll check as soon as I get a new update for the original rg35xx out of the door.