rg35xx-cfw/rg35xx-cfw.github.io #84: eduke3d

benchristey, 2024-01-27:
I've not been able to get eduke32 to work - is this supported?

garland410, 2024-01-28:
Don't want to overstate because my memory is crap and I'm pretty sure I only loaded the title screen back then to test, but I recall this working, at least up to that point a few updates ago.

benchristey, 2024-01-28:
Hmm I was sure I followed the instructions but mine just goes to a black screen then returns to the menu. If you get a sec I’d appreciate you sharing your folder/file structure and the contents of your launcher file.

acmeplus, 2024-01-28:
Just checked and there is a compilation issue with the eduke32 version included in the firmware, so that's why it's not working. It'll be updated in the next version

benchristey, 2024-01-28:
Ah that explains it. Thanks!

benchristey, 2024-02-09:
Its working on the latest release thanks!

closed 2024-02-09 by benchristey