rg35xx-cfw/rg35xx-cfw.github.io #94: Carbon theme??

npaladin2000, 2024-01-31:
Is it just me? I can't find the files for es-carbon anywhere, even though they show up in the theme list. Looking to tweak it for better readibility (it's really meant for bigger screens) but I can't find it.

acmeplus, 2024-01-31:
The files are internal and part of the batocera rootfs partition and not visible in the SHARE partition.

You can download that from the batocera them repo directly: https://github.com/RetroPie/es-theme-carbon

npaladin2000, 2024-01-31:
Thanks but that doesn't seem to be the same es-carbon theme we have in this build. I"ll have to dig around further, but I should be able to get it from EmuELEC if nothing else

acmeplus, 2024-01-31:
Sorry, this it the one actually used in batocera: https://github.com/fabricecaruso/es-theme-carbon