rg35xx-cfw/rg35xx-cfw.github.io #96: Dreamcast performance VS Stock OS

benchristey, 2024-02-01:
The one system I'm finding frustrating is Dreamcast. Crazy Taxi from from the Stock OS was extremely playable but I've not been able to get it to run well at all on Batocera....are there some tweaks I can make at my end?

DutchmanAZ, 2024-02-07:
Runs great for me on the 35XX H. Didn't test extensively but was super playable and vibration was working well. 

What was the specific issue you had?

Lagovas94, 2024-02-13:
I confirm this issue. Games from Dreamcast run perfectly on the stock firmware, but on Batocera, there are severe freezes. Sound stuttering, games run slowly. Especially Resident Evil 2, which doesn't run on the stock firmware.

jdub850000, 2024-02-13:
I'm still getting a freeze coming out of RA with DC titles. As before, FlycastVL and DSP off.
[image]

jdub850000, 2024-02-13:
> I confirm this issue. Games from Dreamcast run perfectly on the stock firmware, but on Batocera, there are severe freezes. Sound stuttering, games run slowly. Especially Resident Evil 2, which doesn't run on the stock firmware. 

I don't get performance issues - it runs great for me - just the freeze when quitting

jdub850000, 2024-02-13:
Did you switch to FlycastVL?

Lagovas94, 2024-02-13:
I switched it, but the game didn't work.  Device Rg35xx plus

TMTwatch, 2025-12-24:
> Any solutions?

If you read the main page for this git you'll see it's been merged into a diff project entirely.