rg35xx-cfw/rg35xx-cfw.github.io #101: Syncthing not working

Northwave, 2024-02-05:
Syncthing seems to be not working at the moment. I added the command 'system.syncthing.enabled=1' to /userdata/system/batocera.conf and also tried with and without the services -> syncthing switch in options. Both didnt work and the port 8384 wasnt shown in netstat. I'm on the  26/01/2024 release

Northwave, 2024-02-05:
Trying to start it manually Produces the following error:

[root@buildroot /userdata/system]# /usr/bin/syncthing -browser-only
2024/02/05 22:05:58 Invalid version string "2b81b0f6";
        does not match regexp ^v\d+\.\d+\.\d+(-[a-z0-9]+)*(\.\d+)*(\+\d+-g[0-9a-f]+)?(-[^\s]+)?$

[root@buildroot /userdata/system]# /usr/bin/syncthing serve --no-browser --no-restart --logflags=0
2024/02/05 22:06:22 Invalid version string "2b81b0f6";
        does not match regexp ^v\d+\.\d+\.\d+(-[a-z0-9]+)*(\.\d+)*(\+\d+-g[0-9a-f]+)?(-[^\s]+)?$

acmeplus, 2024-02-05:
Thanks, yes, it's a build string issue. I've it starting on my development environment, but still synchronization is not working so there's more work to do.

Attached is that test version, simply copy it using adb or ssh to /usr/bin/. Let me know if you get it to work or if it needs more configuration. 

Thanks!

[syncthing.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/14172117/syncthing.zip)

boolemancer, 2024-02-07:
Yeah, this looks like the same as issue #85.

> [syncthing.zip](https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/files/14172117/syncthing.zip)

I tried out this build and it does seem to have fixed the version string, so @Northwave, perhaps you missed something when copying it to your device.

```bash
syncthing --version
syncthing v1.27.3 "Gold Grasshopper" (go1.21.6 linux-arm) builder@github.syncthing.net 2024-01-15 03:45:19 UTC
```

And when overwritten into `/usr/bin/syncthing`, the toggle in the UI does seem to start the service running in the background.

```bash
 ps -A | grep syncthing
 3953 root     /usr/bin/syncthing serve --no-upgrade --no-browser --logflags=2 --logfile=/userdata/system/logs/syncthing.log --home=/userdata/system/configs/syncthing/ --gui-address=http://0.0.0.0:8384
 3961 root     /usr/bin/syncthing serve --no-upgrade --no-browser --logflags=2 --logfile=/userdata/system/logs/syncthing.log --home=/userdata/system/configs/syncthing/ --gui-address=http://0.0.0.0:8384
 ```

Not entirely sure why there are two processes running, but it seems to work fine.

To configure sharing you need to connect to the GUI endpoint (http on port 8384) and set up the connections to whatever other devices you want to sync to. As far as I can tell, this works fine.

Northwave, 2024-02-07:
Yeah something seems to went wrong. will check this later today. 

@boolemancer Did you manually added the 'system.syncthing.enabled=1' into batocera.conf ?

Northwave, 2024-02-07:
looks like it doesnt survive a reboot.... I copied it into with winscp

right before 

[root@buildroot /userdata/system]# syncthing --version
2024/02/07 21:52:18 Invalid version string "2b81b0f6";
        does not match regexp ^v\d+\.\d+\.\d+(-[a-z0-9]+)*(\.\d+)*(\+\d+-g[0-9a-f]+)?(-[^\s]+)?$

and after copy

[root@buildroot /userdata/system]# syncthing --version
syncthing unknown-dev "Fermium Flea" (go1.19.10 linux-arm) ctalon@c31bbf23ae71 2024-01-18 17:50:33 UTC
[root@buildroot /userdata/system]#

also my output for the version is different ?! are you sure you've used the same file from the zip ?

But i was able to run it manually with:  /usr/bin/syncthing serve --no-upgrade --no-browser --logflags=2 --logfile=/userdata/system/logs/syncthing.log --home=/userdata/system/configs/syncthing/ --gui-address=http://0.0.0.0:8384

the trigger in the services settings is always on but reactivating it didnt start the service for me

acmeplus, 2024-02-08:
Unfortunately the binary won't persist for another reboot, I'll included an updated copy in the next release though.

@boolemancer did you manage to actually get the data synchronized with the desktop app? I've not really used Syncthing so I don't know what to expect there.

boolemancer, 2024-02-09:
> I'll included an updated copy in the next release though.

Just updated to the latest build (20240208), and I still see the version string issue with syncthing. Is it safe to assume this was meant to be the *next* next release, or did something go wrong this time?

ArmoMike, 2024-02-14:
> > I'll included an updated copy in the next release though.
> 
> Just updated to the latest build (20240208), and I still see the version string issue with syncthing. Is it safe to assume this was meant to be the _next_ next release, or did something go wrong this time?

Updated to 20240213 and still have the same issue as well

mhbates, 2024-02-24:
Will the fix be included in the next update?

acmeplus, 2024-02-24:
> Will the fix be included in the next update?

I’ve a version ready that requires more testing. I can publish for anyone who wants to give it a try

mhbates, 2024-02-24:
> > Will the fix be included in the next update?
> 
> I’ve a version (that should fix sync thing, sound, and charging) ready that requires more testing. I can publish for anyone who wants to give it a try

I would be happy to help test.

mhbates, 2024-02-26:
Syncthing seems to be fixed with v39.