rg35xx-cfw/rg35xx-cfw.github.io #106: The console does not shutdown fully when you unplug it during charging mode

boolemancer, 2024-02-09:
> Added charger logic. After a cable is connected to the console the console boots and enters charger mode. To continue boot, press the power key once you see the prompt

Once the console has booted into charger mode, I expect it to fully power down if it's unplugged while still in charger mode. Instead it stays booted into charger mode, though the screen does eventually turn off, you can tell it's still powered on because a short tap of the power button will wake it back up to display the charger mode screen (and at least on my unit I can still hear some high pitched electronic whining to indicate it's still on :))

DutchmanAZ, 2024-02-09:
On mine, it literally says Discharging: XX% after I unplug it and it is still in charging mode.  But it is discharging very quickly.  After about 15 mins it had gone from 100 to 91.  I then booted it and the onboard batt percentage was 96.  I think it best to boot and shut down for now.  I just did that.  I will report back about how much seems to discharge when it is fully shut down and not in charging mode.

VulcansMeMoRy, 2024-02-18:
same for me here