rg35xx-cfw/rg35xx-cfw.github.io #109: HDMI unplug causes weird state

jamespharaoh, 2024-02-09:
When I unplug the HDMI cable while playing a NES game (for example) the main menu loads up, but the game seems to keep on playing, with element of both visible on screen. If you can't reproduce this then I can provide more information. This is running on an RG35XX H with the 20240208 firmware. Thanks.

DutchmanAZ, 2024-02-09:
I had same issue when I accidentally switched inputs on my display, turned it off, then turned it back on.  I had weird behavior as jamespharaoh described

lezrule, 2024-03-29:
same here... I got a HDMI switch and the whole system froze if I switch to another channel and switch back. Nothing will work apart from force reboot. (i.e. long hold the power button)