rg35xx-cfw/rg35xx-cfw.github.io #114: SCRAPER NOT WORKING

Keitann0, 2024-02-10:
The Scrapper function does not work at 100% and i been testing all the options and the only one that actually was working stopped working and now everytime i want to scrap from there it displays a error message that says: 

_"{"code":403,"status":"Invalid API key was provided.","remaining_monthly_allowance":0,"allowance_refresh_timer":251414}"._

 And not only that but when i use the other options it does not dowload all the game data for all games (Just dowloads like 40% or less and the rest empty). This happens with NES, SNES and GBA roms.

HansWurstSenior, 2024-02-11:
Works for me with screenscraper. Put in my username and password and it scraped all 2000 roms with no issues.

DutchmanAZ, 2024-02-11:
> The Scrapper function does not work at 100% and i been testing all the options and the only one that actually was working stopped working and now everytime i want to scrap from there it displays a error message that says:
> 
> _"{"code":403,"status":"Invalid API key was provided.","remaining_monthly_allowance":0,"allowance_refresh_timer":251414}"._
> 
> And not only that but when i use the other options it does not dowload all the game data for all games (Just dowloads like 40% or less and the rest empty). This happens with NES, SNES and GBA roms.
> 
> Edit: It still happens even if the rom has no special characters

No offense, but are you hard of reading? It literally names your issue. "Remaining monthly allowance = 0. You need to sign up for an account to have a higher limit. 

But you know you could also just scream in all caps at the single developer working his ass off for all of us...

Keitann0, 2024-02-11:
> No offense, but are you hard of reading? It literally names your issue. "Remaining monthly allowance = 0. You need to sign up for an account to have a higher limit.
>
> But you know you could also just [...]

Ok but then where do i have to log in?. That's not the screenscrapper option is other. 

Who is screaming to nobody?. I think the hard to read one is you making a big drama just be cause i posted this issue title using all caps and you are the one that acts like that means something else. 

He's a solo dev, yes, but he took this like a job and he's accepting donations for doing the job (I already donated him, did you donate him at all?) so i think it's stupid to say: "Hey everybody, stop posting in the issues section even if the OS does not work at its fullest just be cause it's a solo developer" be cause this is the only way he can knew what to do and how to make it better and he is free to read and just ignore and not doing anything about it but as long as he keeps working on the project why is it a bad thing to let him know the bugs of the project when he wants to work on it and correct them?

Keitann0, 2024-02-11:
> Works for me with screenscraper. Put in my username and password and it scraped all 2000 roms with no issues.

What version do you have? i downloaded the newest version and in that version it's where it happens. 

I tried re-booting it again just in case but it's the same, at then end it does not work 100% well.

Keitann0, 2024-02-11:
> No offense, but are you hard of reading? It literally names your issue. "Remaining monthly allowance = 0. You need to sign up for an account to have a higher limit.
>
> But you know you could also just [...]

Let me correct that for you, there you go, now the tittle is not in all caps, do you feel better?. Now im not sreaming? or is it another interpretation you are doing to my post that offends you in any other way?.

Correct me if im wrong but i think this is Github and if he uploaded this here and allowed people to write a issue in the issue section it must be bc he wanted to know if there are any issues with the OS he's developing so he can solve those problems and make it better, right?

Now, we're on Github so if you just come to make a big drama over an all caps tittle that you interpreted as "Me screaming to the only dev" just go somewhere else.

Billychad89, 2024-04-29:
> Works for me with screenscraper. Put in my username and password and it scraped all 2000 roms with no issues.

Where do you input username and password? I can't find it in the UI?