rg35xx-cfw/rg35xx-cfw.github.io #115: Wifi not connecting to wpa+wpa3

HansWurstSenior, 2024-02-10:
Wifi is not connecting to my Fritz!Nox 5530 Fiber router, if security is set to "wpa2 + wpa3".
If I set the Router to "wpa2 (ccmp)", it works.

It's no problem on stock os or on any other wifi device I use.

HansWurstSenior, 2024-02-10:
For what I read (I'm not an expert), with the wpa2+wpa3 transition mode enabled, the Router also uses "PMF", even with wpa2 connections.
https://www.wi-fi.org/beacon/philipp-ebbecke/protected-management-frames-enhance-wi-fi-network-security

psavvides, 2024-02-13:
Same experience, device would not connect under WPA2/WPA3 mode even with PMF set to Optional on the AP.  Setting WPA2-only would work fine, and irrespective of wifi key length (eg not affected by the stock firmware issue with max 16 char wifi keys)

6jarjar6, 2024-02-13:
Also had this problem, enabled IoT AP on router. Allowed me to connect.