rg35xx-cfw/rg35xx-cfw.github.io #117: 32 bit toolchain only?

crudelios, 2024-02-10:
Hi, 

I noticed that the provided toolchain creates 32-bit ARM7 binaries (at least when using the provided cmake toolchain file), but both rg35xx plus and h have arm64 capable processors.

Is there a particular reason for that? Does batocera only run in 32 bits?

Thanks for all the hard work!

crudelios, 2024-02-16:
I'm guessing the reason is that the handheld only has 1gb of ram, so it doesn't make much sense to use 64 bit binaries.

Anyway, I'm closing this!

closed 2024-02-16 by crudelios

acmeplus, 2024-02-17:
Sorry, I missed this issue. The reason is not the memory, but the lack of a 64bit gpu blob that uses the fbdev and works on this architecture. There's little point having a 64bit toolchain if you want to compile graphical applications (emulators, etc.). Right now the system is using libmali g31 fbdev version for 32bits, so that's the main driver. 

I've done some preliminary work to get a new kernel compiled, but not yet a mainline one that will require significant work. So it's really unlikely that we will have a working 64bit version anytime soon.

crudelios, 2024-02-17:
Thank you very much for taking the time to answer!

It's a shame there aren't proper working drivers for this device. I also noticed that the original firmware also only compiles 32 bit binaries by default and can't run 64 bit binaries, so this really is a deeper issue.

Keep up the fantastic work!

KatyushaScarlet, 2024-03-13:
> Sorry, I missed this issue. The reason is not the memory, but the lack of a 64bit gpu blob that uses the fbdev and works on this architecture. There's little point having a 64bit toolchain if you [...]

Hi, I found that armbian has already ports for H616 and H618. Is it possible to reference their work? After all, H616, H618, H700 have very similar specs.