rg35xx-cfw/rg35xx-cfw.github.io #121: 3.5mm audio may not be working

DutchmanAZ, 2024-02-11:
Need someone else to test 3.5mm audio jack and see if it's outputting. 

I tried on both my plus and my H and it stops the speakers but doesn't actually put any sound through my headphones. 

I have seen some posts with hardware issues relating to solder joints, so if someone else can confirm that would be great.

acmeplus, 2024-02-11:
It has not been really tested so it does not surprise me they it does not work. Will have a look at it.

jdub850000, 2024-02-12:
3.5mm works good here  - in-menu and with a psx test.

acmeplus, 2024-02-12:
Yeah.  Tested it yesterday and it was working as expected.

hdweiss, 2024-02-12:
@DutchmanAZ Have you tried turning up the volume? I thought it didn't work, until I did that

DutchmanAZ, 2024-02-13:
Yeah I think it is either my headphones or the units themselves. Thanks for checking guys! I'll troubleshoot on my end.