rg35xx-cfw/rg35xx-cfw.github.io #122: Some issue found in latest build (20240211)

eltoffer, 2024-02-11:
First of all thanks to ACMEPLUS for the awesome job done, I now how hard could it be to get to this point so fast, so thanks for your time and support

Some bugs i've found since the previous 3/4 releases until now

-In retroarch, assets won't be saved after shut down the device
-N64 emulators doesn't map the joystick (tested on Mario Kart 64, Mario 64)
-In Nintendo DS emulator can't access to settings, so saved games cannot be achieved
-Cannot connect to a WiFi network with uses the same SSID for 2.4gHz / 5.0 gHz bands (AKA "smart connection")

Thanks again pal

HansWurstSenior, 2024-02-11:
Are you shure, that the 2.4/5.0ghz is what preventsnit from connecting? My tests show, that it's related to wpa3. If I disable that and set my router to wpa2, it connects (even with the 2.4/5.0ghz enabled.

I made up this issue: https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/issues/115

eltoffer, 2024-02-12:
> Are you shure, that the 2.4/5.0ghz is what preventsnit from connecting? My tests show, that it's related to wpa3. If I disable that and set my router to wpa2, it connects (even with the 2.4/5.0ghz enabled.
> 
> I made up this issue: #115

My router has no WPA3 support so im definitely sure it is not my issue, I disabled the 5gHz band and the device connects properly

HansWurstSenior, 2024-02-12:
Okay, I will test some more settings and report back. Strange...you have no problems on stock os with these settings, right?

eltoffer, 2024-02-12:
not at all, I could properly connect to my network, thanks

eltoffer, 2024-02-12:
> Okay, I will test some more settings and report back. Strange...you have no problems on stock os with these settings, right?

if you suggest any special settings for my router I could try as well and report back

acmeplus, 2024-02-12:
If you have PMF set up. Try to disable it. It’s not a solution but may be a temporary workaround.

eltoffer, 2024-02-12:
I think I've found the cause, I have two routers with the same SSID, setting two differences SSID names the device properly works with both routers, its strange since on stockOS it didn't happen 

well maybe it helps you somehow

ioiomi1, 2024-02-12:
> Are you shure, that the 2.4/5.0ghz is what preventsnit from connecting? My tests show, that it's related to wpa3. If I disable that and set my router to wpa2, it connects (even with the 2.4/5.0ghz enabled.
> 
> I made up this issue: #115

I've been using wpa2, but I often can't connect. I have to try again a few times.

AndreaCCIE, 2024-02-12:
> Okay, I will test some more settings and report back. Strange...you have no problems on stock os with these settings, right?

Same here, I have to join a 2.4Ghz only SSID otherwise it doesn't connect.

RedCider, 2024-02-12:
For me it does not work with WPA2+3 enabled, as soon as I switch to WPA2 only it works. My WiFi is set up as 2.4 GHz and 5Ghz with same SSID. The device connects with 2,4 GHz.

james-mccloud, 2024-02-12:
I also found some issues in DS Controls. For example, pressing the A button is like pressing A + ->, pressing R is like pressing R + A, and pressing L is like pressing L + B. At least.

eltoffer, 2024-02-12:
> I also found some issues in DS Controls playing on RG35XX H . For example, pressing the A button is like pressing A + ->, pressing R is like pressing R + A, and pressing L is like pressing L + B. At least.

Yeah, some mappingg issues as well, I think when the DS menu comes available that could be fixed, the same for N64

james-mccloud, 2024-02-13:
NDS controls works ok on 20240213. 
good job @acmeplus 
Thank you. 👏🏻👏🏻👏🏻

eltoffer, 2024-02-13:
> NDS controls works ok on 20240213. good job @acmeplus Thank you. 👏🏻👏🏻👏🏻

just updated the thread