rg35xx-cfw/rg35xx-cfw.github.io #124: Missing source code ?

6jarjar6, 2024-02-13:
Maybe I am just missing something, but why isn't the source code on github, and instead uploaded by zip?

Thanks,
6jarjar6

loki666, 2024-02-29:
indeed, please publish your batocera fork
ideally it should be merged back with upstream so we can get official support / update from batocera

acmeplus, 2024-02-29:
Everything is there. It’s at the organization level and not at the project level. The releases are not tied to the repos because it’s a combination of different repositories. 

Have a look at:
https://github.com/rg35xx-cfw/batocera.linux/tree/rg35xx-plus-wip
And 
https://github.com/rg35xx-cfw/buildroot/tree/rg35xx-plus-wip

acmeplus, 2024-02-29:
And @loki666 about merging back into batocera that won’t happen until much later if anything. Batocera team needs the devices to be able to maintain them and they don’t have these devices. I’ve already contributed back many fonsole support back to batocera main (a12, a13, ps5000, ps7000, unico R1) but some devices require too many changes that break compatibility with the main batocera repo.