rg35xx-cfw/rg35xx-cfw.github.io #126: NES roms not working

pierre-gerard, 2024-02-13:
My ROM files are .nes in the NES folder. They were working with stock OS. I get a black screen and back to the game list. I'ce tried with everything AUTO anb then, with every core.

DutchmanAZ, 2024-02-13:
Look at the text file in the NES folder and make sure your file extensions are supported. Pretty sure that is your issue

pierre-gerard, 2024-02-13:
Thanks for the answer but I have regular .nes files and they are supported according to the text file. Since the games are listed, it has to be.

I nevertheless have found one setting which works : MESEN core when the Console Region option is set to NTSC. Otherwise it doesn't work.

Since the option doesn't come with other cores, I cannot use them.

With MESEN, the sound is awful. Almost distorted. I hope that there is a solution to ise other cores.