rg35xx-cfw/rg35xx-cfw.github.io #128: Weird issue when trying to disable threaded video in RA

DutchmanAZ, 2024-02-13:
Having a weird thing happen when I go into the RA menu for any of the SNES emulators. I get to the setting to toggle threaded video off, and it maybe toggles, but you can't tell, and then attempts to save the config file automatically? But in doing so brings you to a different looking RA menu that I cannot do ANYTHING in. Doesn't recognize any inputs. The only option is to reset the unit. When I go back in, it is still set to on. 

If anyone could confirm it isn't just me that would be great. The threaded video setting is in the main settings (you have to hit back when you first enter the RA menu), then video, then output.  If you are noticing issues with stuttering in SNES high action games, this is the thing we need to turn off to fix that.

eltoffer, 2024-02-15:
doing so in my unit (latest update 20240213) the RA just crashed

DutchmanAZ, 2024-02-16:
Acmeplus (the man, the myth, the legend) kindly pointed out that this can easily be done in the config files:

"You can always override default variables, if video_thread is a problem, edit [SHARE]/system/batocera.conf file and at the end of that file add:

global.retroarch.video_threaded = "false" 

I confirm this works for me and will solve some of the perceived lag or fps issues in most RetroArch emulators.