rg35xx-cfw/rg35xx-cfw.github.io #129: Can't update

fantomac, 2024-02-13:
Hi, First, thanks for your work ❤️
I deleted the file boot/batocera from the first partition of the SDCARD (BATOCERA) but when I want to copy the new batocera file, as the new file is heavier than the old one, there is not enough space. 
Did I miss a step?

Have a nice day !

DutchmanAZ, 2024-02-13:
What system are you using to do this? Mac or PC? 

If on Mac, try emptying your trash after you scrap the boot/Batocera file.

Me and others have not had this issue when trying to upgrade, so it is likely something in your system or process.

acmeplus, 2024-02-13:
Yeah, not emptying the Trash in a Mac may cause that problem.

fantomac, 2024-02-13:
Ok, my bad.
I saw the file (2.9go) much bigger than the partition (1.29go).
I'll try next time (I'm on Mac).
Thanks