rg35xx-cfw/rg35xx-cfw.github.io #130: `batocera-save-overlay` error

petertriho, 2024-02-14:
Trying to modify the system using https://wiki.batocera.org/modify_the_system_while_it_s_running

```
[root@buildroot /userdata/system]# batocera-save-overlay
Making /boot writable...
Creating an overlay file on the /boot filesystem...
1+0 records in
1+0 records out
104857600 bytes (105 MB, 100 MiB) copied, 3.4461 s, 30.4 MB/s
Formating the overlay file in ext4...
mke2fs 1.46.5 (30-Dec-2021)
Creating filesystem with 102400 1k blocks and 25584 inodes
Filesystem UUID: 5849e870-0841-40c7-97f1-28500af5cd8f
Superblock backups stored on blocks:
        8193, 24577, 40961, 57345, 73729

Allocating group tables: done
Writing inode tables: done
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: done

Mounting the overlay file...
mount: can't setup loop device: Inappropriate ioctl for device
```

Seems like creating the overlay file works fine (I can see the overlay file at `/boot/boot/overlay`) but changes made to the system are not applied.

Changes I'm trying to make is to change the save state/file directories by modifying https://github.com/batocera-linux/batocera.linux/blob/e82c95298dab07be4fb8a6cbe58920529d4d5b42/package/batocera/core/batocera-configgen/configgen/configgen/generators/libretro/libretroConfig.py#L197

```
retroarchConfig['savestate_directory'] = "/userdata/states/"
retroarchConfig['savefile_directory'] = "/userdata/saves/"
```

acmeplus, 2024-02-14:
Unfortunately that does not work, the kernel has issues mounting the overlayfs. batocera-save-overlay works correctly, but mounting does not working. I've been investigating that issue but haven't solved it yet and it's not clear whether it will be solved unless we change the kernel, and that's not trivial.

petertriho, 2024-02-14:
Ah ok, that's unfortunate. I don't like how batocera handles save files/states but there doesn't seem like there's another way to split them up and remove the system name so that I can sync saves with my other devices

acmeplus, 2024-02-14:
The more cumbersome way to do quick persistent modifications is in linux as follows: 
* Mount the BATOCERA partition (actually the third one on linux) of the sdcard on a linux system (e.g. mount /dev/sdh3 /mnt/batocera)
* Change directory to the boot folder (e.g. cd /mnt/batocera/boot)
* Create a folder to mount that, e.g. mkdir /tmp/rootfs
* Mount batocera there, mount batocera /tmp/rootfs
* cd to /tmp/rootfs and make your changes (you will typically need sudo or root permissions)
* umount /tmp/rootfs 

Then the changes will be there when you boot with that card.

In any case I plan to eventually solve the overlay issue, given the kernel version it should be possible to get it to work properly.

petertriho, 2024-02-14:
Seems like my device doesn't have `/dev/sdh3`. Tried both `/dev/mmcblk0p3` and `/dev/disk/by-label/BATOCERA` mounting to `/mnt/batocera` which worked but trying to `mount batocera /tmp/rootfs` gives the same error `mount: can't setup loop device: Inappropriate ioctl for device`