rg35xx-cfw/rg35xx-cfw.github.io #131: NES lag

PsychoNewt, 2024-02-14:
SNES plays beautifully. Some NES games come almost to a complete halt at times depending on the number of on-screen elements (seemingly). Also, there are FPS issues on NES where there are none on SNES.

DutchmanAZ, 2024-02-15:
Odd I am noticing something similar in SNES. What games are you testing?

PsychoNewt, 2024-02-15:
The NES lag I referenced is in super mario bros 2. It is playable but I found myself having to fast forward through some areas where the game had too many enemies on screen and the game was barely progressing. I only played super mario world in SNES and didn't encounter issues in the few levels I played.

DutchmanAZ, 2024-02-16:
I confirmed I am seeing it in Kirby's Adventure on the NES as well. In my experience this is related to threaded video setting in RetroArch. It is currently defaulted to on and we want it off, unfortunately turning it off doesn't seem to be possible at the moment. I have another issue opened identifying the inability to turn off threaded video. Once that is fixed, everything will run much smoother I think, as many FOS as stock, if not more.

acmeplus, 2024-02-16:
You can always override default variables, if video_thread is a problem, edit ``[SHARE]/system/batocera.conf`` file and at the end of that file add:
```ini
global.retroarch.video_threaded = "false"
```

DutchmanAZ, 2024-02-16:
> You can always override default variables, if video_thread is a problem, edit `[SHARE]/system/batocera.conf` file and at the end of that file add:
> 
> ```ini
> global.retroarch.video_threaded = "false"
> ```

Duh! I had totally forgotten about adjusting config files. I'll pop in and give it a go. Thanks for the help and all your work!!!!

DutchmanAZ, 2024-02-16:
> You can always override default variables, if video_thread is a problem, edit `[SHARE]/system/batocera.conf` file and at the end of that file add:
> 
> ```ini
> global.retroarch.video_threaded = "false"
> ```

They totally worked for disabling threaded video! Contra Alien Wars is running much better now on SNES as well as R Type 3. I tried Kirby's Adventure on NES, and noticed slowdown still, but looked it up and that is just how it ran on NES and just runs that way no matter what

PsychoNewt, 2024-02-17:
> You can always override default variables, if video_thread is a problem, edit `[SHARE]/system/batocera.conf` file and at the end of that file add:
> 
> ```ini
> global.retroarch.video_threaded = "false"
> ```

If I'm running Windows 10 with no EXT4 support, what's the easiest way to access that configuration file?

acmeplus, 2024-02-17:
> If I'm running Windows 10 with no EXT4 support, what's the easiest way to access that configuration file?

The SHARE partition is exfat that works in all systems. Just mount the sdcard on your PC and open that partition

PsychoNewt, 2024-02-17:
> The SHARE partition is exfat that works in all systems. Just mount the sdcard on your PC and open that partition

THANKS!!! To anyone who was having the same problem as me (windows not recognizing any readable partitions on their card) you can go to disk management in windows and assign the "share" partition a drive letter manually. I thought I couldn't see it because it was EXT4. DOH!

PsychoNewt, 2024-02-17:
I'm calling this one closed, after adding the line to the config, NES/SNES seem to be running as well as they did on my RG35XX. Thanks! 👍

closed 2024-02-17 by PsychoNewt