rg35xx-cfw/rg35xx-cfw.github.io #132: Battery drain when system is in shutdown

deandcd, 2024-02-14:
While system is in shutdown, there is a noticeable battery drain.

Simb51, 2024-02-15:
same too, loss of 4% per hour on average

A-Bony-Mouse, 2024-02-15:
I am also experiencing this issue. Does not seem to make a difference if I shut it down through the menu or turn it off by pressing and holding the power button until the power indicator LED turns off. Has anyone tested it compared to the Stock OS?

chris101822, 2024-02-16:
Also experiencing the same issue. Lost 6% being powered off for half an hour. Lost only 1% being on for half an hour. I just switched from Stock OS. Didn't have that issue before.

acmeplus, 2024-02-16:
There's a know issue that affects the charging logic when the device has been connected to the power and the power has been disconnected. Instead of automatically shutting down, it stays in suspend mode and that's the reason it consumes power.

A-Bony-Mouse, 2024-02-16:
I did some more testing and it looks like I was a little hasty in my assessment, turning it off using the power button does work. Turned it off like that and when I booted it 2.5 hours later the battery level was the same as it was before I powered it off. 

I did notice the battery drain after I had charged the device, but I unplugged and shutdown through the menu. I'll keep that in mind for next time because it did look like it was off, no LEDs were on or anything.

acmeplus, 2024-02-16:
Yeah the important bit right (until I fix the charger code) is to make sure the console is not charging when you really want to power it off.

deandcd, 2024-02-17:
Did some testing as well. If i fully charge the device and turn the device off through the menu or with the button, it will drain the battery. When on lower levels of battery, there is no drainage. Strange.

Keitann0, 2024-02-19:
> While system is in shutdown, there is a noticeable battery drain.

Same here, it's crazy how i can shut it down with 90% of battery and at the next day it just have 16%.

AndreaCCIE, 2024-02-22:
Same problem, it dropped 20% overnight. I did shutdown the device, charged it to 100%, and disconnected the device from the charger. All LEDs were off

jdub850000, 2024-02-22:
I've run a test with my RG-H and have found the battery to only drop about 1% per day over the last 2-3 days, so the drain issue isn't across the board.

AndreaCCIE, 2024-02-22:
> I've run a test with my RG-H and have found the battery to only drop about 1% per day over the last 2-3 days, so the drain issue isn't across the board.

Mine is an H version as well. Some processes don’t shut properly

jamesaslett1985, 2024-02-22:
Yeah, so when it's properly off (eg: pressing the power button doesn't prompt you to press again to boot) there is definitely substantial battery drain. I just noticed approximately 10% loss in 2 hours. And that's when I shut the system down via the menu. Also, charging the console appears to 'wake' it out of shutdown, because once you have connected it to power, you can press the power button and see the 'press again to boot' screen, like it's in sleep).

waffledork, 2024-02-28:
@acmeplus - with the latest update (Feb 20), the RG35XX+ automatically shuts down after unplugging the charger - great job with this! But even while unplugged and off, the battery still drains quite a bit. It went from 100% to 90% overnight when it was just sitting in its case. 

Also, would it be possible to keep the RG35XX+ from turning on when it's plugged in? This way, you won't have to shut it down afterwards and it would charge faster because it wouldn't also need to provide power to keep the device on.

rebeltaz, 2024-03-22:
I noticed that my rg35xx H was dropping about 20% over night, just sitting there. I saw here mentioned that this will happen if you shut it down with it charging, but that is not the case for me. Without it being on charge, or having been on charge at all over the course of several ons and offs, I got to the menu and do either Shutdown or Fast Shutdown. There are no LEDs on or any other sign of power. Over the course of six hours, though, it will drop about 20%. I thought it might be that I forgot to turn wifi off, so I did that and still get the power drain. I noticed that there is no way (that I could find) other than Airplane Mode to turn bluetooth off, but I haven't tried that yet. I did turn the rg off, take the batocera SD out, put the OEM SD in, boot back up and shut back down there. After six hours, I put the batocera SD back in and booted back up... no battery loss at all. Since there is already a thread for it, I didn't want to open a new issue... not sure of the protocol on that, so...  🤷‍♂️

seppeconinx, 2024-04-02:
Are there any more updates concerning this issue? I am using the v40 beta and the battery drains like crazy after I choose "shutdown".. I am losing like 10% per hour right now and have already restarted and shut it down again

seppeconinx, 2024-04-02:
@acmeplus Are there any more updates concerning this issue? I am using the v40 beta and the battery drains like crazy after I choose "shutdown".. I am losing like 10% per hour right now and have already restarted and shut it down again

rebeltaz, 2024-04-02:
> @acmeplus Are there any more updates concerning this issue? I am using the v40 beta and the battery drains like crazy after I choose "shutdown".. I am losing like 10% per hour right now and have already restarted and shut it down again. And like someone else mentioned, it seems to drain more if your battery is quite high in %

What you can do, until an update is released, is - with the device on - press and hold the power button until the LED goes our. This fully turns it off and it doesn't drain the battery. I can leave it overnight in this state without losing any charge at all.

Mordenheim, 2024-04-06:
Charged it fully yesterday afternoon, booted it up, did a full shutdown, (saving metadata) when I tried to turn it on at about 10:00 this morning it brought up the boot screen for a second, then powered off. Plugging it in showed a battery level of 0%.

waffledork, 2024-04-06:
@Mordenheim - this seems really extreme. Did you happen to do the full shutdown while it was plugged in? If so, I think you ran into this issue: https://github.com/rg35xx-cfw/rg35xx-cfw.github.io/issues/219#issue-2228032351

If not, you might need to replace the battery altogether.

Mordenheim, 2024-04-06:
@waffledork DId the full shutdown after unplugging.  Currently made a backup of my 2nd micro sd card and moved to modified stock os MOD3 to see if the problem persists.

uzumak11, 2024-04-06:
Ever since I've switched to ext4, I don't have battery drain. Switched the device on after 3days, and still had 90% charge. 

Very anecdotal, but just wondering if this is the case for others.

rupokify, 2024-04-07:
> Ever since I've switched to ext4, I don't have battery drain. Switched the device on after 3days, and still had 90% charge.
> 
> Very anecdotal, but just wondering if this is the case for others.

> ext4 is the default userdata partition filesystem [https://wiki.batocera.org/batocera.linux_architecture] 
@uzumak11 What other change did you do specifically? I'm running ext4 default userdata partition and I'm still facing serious battery drain overnight.

uzumak11, 2024-04-07:
Isn't exfat default on here for our device?

rupokify, 2024-04-07:
@uzumak11 You are right. And it seems like that also worked for me. Great find.

uzumak11, 2024-04-07:
It's so weird. And I couldn't even tell you why it works.

Perhaps someone else can do a deep dive. 

I've had SD2 on ext4 for a while now (since v40 released), and it has greatly negated battery drain during shutdown.

rupokify, 2024-04-07:
I'm wondering why v40 would format that partition in exFAT while their official documentation says ext4 is the default userdata partition filesystem. Let's see if this works for other users too.

acmeplus, 2024-04-07:
The reason the filesystem was changed to ExFAT is because many users were always getting back saying they couldn't access the SHARE partition, because it was in EXT4 and they didn't have tools to access it. So in the end it was changed to that format.

The battery drain issue is related to filesystem corruption, so it makes sense that switching to ext4 mitigates that problem. The last version of the firmware (20240306) has an issue with the S03checkfat init script that may explain why there are more issues with FAT32 or ExFAT.

For the next version of the firmware I may actually switch back to ext4 and add scripts or instructions for users to reformat the SHARE partition with a big disclaimer :)

saulob, 2024-04-07:
> The reason the filesystem was changed to ExFAT is because many users were always getting back saying they couldn't access the SHARE partition, because it was in EXT4 and they didn't have tools to [...]

Strange bug indeed. So, to fix it, which partition do we need to change to ext4? The system one or just the ROMs (share) partition? 

And will we lose everything? Do we need to reformat, right? Oh, that's a bummer. 

And if we use ext4 on share, how will we, on Windows, copy the ROMs? What a weird bug, what a frustrating situation, you know

uzumak11, 2024-04-07:
> Strange bug indeed. So, to fix it, which partition do we need to change to ext4? The system one or just the ROMs (share) partition?
>
>
>
> And will we lose everything? Do we need to reformat, right? [...]

Roms partition, the one called share. 

You will lose everything, it's a format essentially. Create a backup and then move your files back.

Windows cannot access ext4. If you need to transfer files wired, you need access to linux on your computer. 

Wireless, you can use network share on your windows pc to access the system