rg35xx-cfw/rg35xx-cfw.github.io #133: Gamepad 8BitDo M30 does not connect to Batocera

AlexLeoni, 2024-02-14:
Good afternoon, is it possible to add support for the 8BitDo M30 gamepad to Batocera? Doesn't connect :(

TrilIiard, 2024-02-15:
Same for the 8BitDo Pro 2

TrilIiard, 2024-02-16:
Yes, its showing "connecting to 8BitDo Pro 2" then "trusting 8BitDo Pro 2" and then "connection failed"

DutchmanAZ, 2024-02-16:
After the initial pairing, even when it says failed, try rebooting. Once you do that, don't try to re-pair it. Just turn it on. it usually just pairs right up no problem using an 8bitdo Lite 2. 

I think the Bluetooth in these units isn't the best.

TrilIiard, 2024-02-16:
Correct, that works. But it works fine on Stock OS with two Controllers at the same time so the Hardware is not the problem here.

AlexLeoni, 2024-02-17:
On the stock firmware 8BitDo M30 connects and works perfectly, without errors.

AlexLeoni, 2024-02-17:
> After the initial pairing, even when it says failed, try rebooting. Once you do that, don't try to re-pair it. Just turn it on. it usually just pairs right up no problem using an 8bitdo Lite 2.

This connection option doesn't work :(