rg35xx-cfw/rg35xx-cfw.github.io #134: Can't connect to syncthing

ArmoMike, 2024-02-14:
Hello I am on the latest version (20240213) and  i have syncthing activated in the settings . When trying to connect to the config page through my browser with (ipadress:8384) it fails to reach the site each time.

0chroma, 2024-02-17:
looked into this, it's due to a bad version string:
```
[root@buildroot /userdata/system]# /usr/bin/syncthing -browser-only
2024/02/16 22:18:10 Invalid version string "2b81b0f6-dirty-rg35xx-plus-wip";
	does not match regexp ^v\d+\.\d+\.\d+(-[a-z0-9]+)*(\.\d+)*(\+\d+-g[0-9a-f]+)?(-[^\s]+)?$
```

ArmoMike, 2024-02-17:
Great thanks , i'll give that a try ! Also does anybody know how I can save custom controller configs per system? As of right now the controls reset whenever i close a game. I've set the controls in the bacotera settings but for some reason some of my emulators won't use them.

benpage, 2024-02-19:
the issue is the 2b81b0f6 doesn't match the regex part which should be in the format v0.0.0.0

I manually edited the file using an elf editor (xelfviewer) and then ran the executable. It then automatically updated itself to the latest version.

mhbates, 2024-02-26:
Syncthing seems to be fixed with v39.

seanstardev, 2024-02-27:
> Syncthing seems to be fixed with v39.

How did you manage this? I couldn't connect via http://192.168.0.??/8384 after following the Batocera wiki page. Thanks.

0chroma, 2024-02-27:
you just enable it under system settings now, should work out of the box

seanstardev, 2024-02-27:
> you just enable it under system settings now, should work out of the box

Thank you very much. Took a couple more resets but I'm there. Really appreciate your time.

closed 2024-02-28 by ArmoMike